sinara-hw/Stabilizer #1: [RFC] New EEM: Stabilizer (Microcontroller Servo)

dtcallcock, 2018-11-15:
A fresh issue for one of the two boards proposed in [[RFC] Sinara Servo #16](https://github.com/sinara-hw/meta/issues/16).

This would be a lower performance board with the feedback loop running on a microcontroller. 

- ~MSPS DACs/ADCs and ~us latency
- Ethernet for standalone operation
- Microcontroller/ethernet layout could be copied from proposed [Humpback](https://github.com/sinara-hw/meta/issues/5) target like [ST Nucleo 144](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F429ZI/497-16280-ND/5806777) to reduce risk and share development costs.
- One EEM to allow a Kasli to control lock parameters if desired _or_ to connect a DIO EEM.
- Low cost, of order $100/ch

This board could also be used as a general lab logger/controller to replace [LabJacks](https://labjack.com/products/t7).

dtcallcock, 2018-11-15:
It looks like the microcontroller ADCs will need an external reference.  Either its own 2.5V chip or divide down by 2 the external DAC's 4.096V reference?

My front panel preference would be for SMA but this is an across the board preference of mine, not specific to this board.

> PGIA is AD8250 (unless @dtcallcock objects strongly

Given it's so easy to swap out, I'm not too bothered.

> I'm less convinced about having "downstream" EEMs

Would it be simple to make it so that the single EEM can be upstream or downstream?  If it doesn't involve any real costs or compromises I think it's worth doing.  

Otherwise this sounds like a good plan to me.

gkasprow, 2018-11-15:
@hartytp I did pin assignments for STM32:
- 4 individual SPI blocks for ADC and DACs
- 2x SPI slave or  master to EEM
- one UART (GPIO pins)
- Ethernet
- timer outputs (digital IOs PWM)
- 2 DACs
- 6 Diff ADC channels, 
- 1 ADC trigger (GPIO pin)
- 2 I2C 
- QSPI (GPIO pins)
- USB device
- JTAG
- several IOs (GPIO pins, EEM)

The question is: what special purpose pins do you want to have connected to GPIO and EEM?
Every pin can be interupt, there are ADC and DAC triggers which I already connected. Maybe some timer capture inputs? The CPU has very rich peripherals and plenty of options.
Here is CubeMX config file, if you want to look at it.
[STM32_config.zip](https://github.com/sinara-hw/Stabilizer/files/2587352/STM32_config.zip)

gkasprow, 2018-11-16:
@hartytp if we want to use termination indicator LEDs for input channels we have to switch to SMA or 8 HP panel. With BNCs I won't fit LEDs between them. I could use whole-metal BNCs which have narrower footprint, but they are not isolated and big washers would be needed.

hartytp, 2018-11-16:
>It looks like the microcontroller ADCs will need an external reference. Either its own 2.5V chip or divide down by 2 the external DAC's 4.096V reference?

Probably easiest to just add a cheap 2V5 reference rather than try to route the low noise DAC reference across the board to the microprocessor (would need a reference buffer anyway to prevent the microprocessor from adding noise to the reference). But, I'm not fussed about that. Whatever @gkasprow thinks is easiest to do.

> My front panel preference would be for SMA but this is an across the board preference of mine, not specific to this board.

For slow stuff I still like BNC. Quicker to do and undo and easy to make one's own cables to the right length for a given application. But, this is a weak preference and I'd be happy with SMA if there is a good reason for it, such as FP space.

> Given it's so easy to swap out, I'm not too bothered.

Cool. Let's go for the 8250 as default in the first revision then. I've found on Sampler that it is quite annoying how many bits of resolution one looses for signals around 2V. Having that extra factor of 5 can make a big difference.

> Would it be simple to make it so that the single EEM can be upstream or downstream? If it doesn't involve any real costs or compromises I think it's worth doing.

- @gkasprow do we need a jumper or something on the EEM connector 12V rail to prevent damage if a user connects both PoE and EEM at the same time?
- assuming the LVDS drivers are bi-directional, AFAICT the only hardware difference between upstream and downstream EEMs is the 3V3MP and the I2C. We'd need a jumper to allow the 3V3MP to be driven by stabilizer instead of Kasli, same for the I2C switch. What do you think @gkasprow?

> @hartytp if we want to use termination indicator LEDs for input channels we have to switch to SMA or 8 HP panel. With BNCs I won't fit LEDs between them.

SMAs are fine IMHO.

If anyone objects to using SMAs, I'd also be okay with removing the termination resistors. I'm not comfortable about using termination resistors without a front panel LED, since it introduces too much room for confusion.

> I could use whole-metal BNCs which have narrower footprint, but they are not isolated and big washers would be needed.

For the ADCs we do need isolation, so either isolated BNC or SMAs + washers is essential. For the DACs, isolation is not so important.

> The question is: what special purpose pins do you want to have connected to GPIO and EEM?

I'll have a think about that. @cjbe @jordens @klickverbot and others, any opinions?

hartytp, 2018-11-16:
> The question is: what special purpose pins do you want to have connected to GPIO and EEM?

For the EEMs it would be nice to have:
- An SPI bus
- One or two time outputs (if there is room)
- The rest just GPIO is fine

For the digital header:
- An SPI bus maybe QSPI?
- I2C
- A few timer outputs/PWM pins
- UART
- ADC trigger
- some gpio

AFAICT, that's basically what you sketched out above. So that seems good to me. Thanks!

hartytp, 2018-11-16:
@gkasprow thinking about this further, I'm beginning to think that we should include the trigger/integrator enable/hold TTL inputs on the front panel. i.e. we should have: 2xanalog in; 2xanalog out; 2xdigital in.

My reason for this is that integrator hold is such a common/useful feature that it would be very annoying to require an external breakout board to access it.

I think that pushes us to use an 8HP board with 3x2 stacked BNCs. I'm fine with that.

The only downside I can see from that (apart from using up more rack space, which is fine IMHO) is that it makes it harder to mount custom AFE daughter cards directly to the PCB (using the 100mil analog/digital headers as board-board connectors). @gkasprow are there any board to board connectors we can use for 8HP panels?

@cjbe thoughts?

gkasprow, 2018-11-16:
why we cannot connect potential AFE with IDC cable?
Such long 100mils headers are available, I used it several times.

hartytp, 2018-11-16:
> why we cannot connect potential AFE with IDC cable?

Well, it would need three IDC cables (main analog IO, digital IO, and probably aux analog IO). But, yes, we can do that. I think @cjbe is hoping to screw the two boards directly together without using IDCs to make things a bit simpler/reduce the amount of wiring (i.e. replace wiring with fixed board-board connectors).

If long 100mil headers are easy to get then that works fine for me. How would that work? Do we have female headers on Stabilizer and long pins on the AFE? Or, do we stick with the usual male headers on Stabilizer and use long female headers on the AFE? Or is there some adapter we can place between the two boards?

cjbe, 2018-11-16:
My preference would be to stick to a 4HP board. I don't think the digital inputs will be used by most people, so I think we should drop these from the front panel. I would also prefer SMA connectors if it means we can get down to 4HP.

This leaves on the front panel USB, ethernet, 4x SMA, and some status LEDs - this should fit.

I would like to be able to mount a piggy-back board with its own front panel if possible (to avoid a mess of IDC cables and potential noise coupling depending on how the cables lie). To do this we would need to find connectors that give a 4HP spacing between the main board and the piggy-back board, as well as having a mounting hole pattern on the main board.

hartytp, 2018-11-16:
> My preference would be to stick to a 4HP board. I don't think the digital inputs will be used by most people, so I think we should drop these from the front panel. I would also prefer SMA connectors [...]

Do you think we have room for 4xSMA (with isolation washers), micro USB, ethernet, some LEDs and 2xMMCX (for the TTLs)? That would be great.

gkasprow, 2018-11-16:
We can have female sockets on both sides and place pins between.

gkasprow, 2018-11-16:
What if we make 4HP with SMAs and simple 4HP AFE with 5 SMAs with 2 digital IOs and 3 slow analog signals? Or simply use IDC-BNC but we make IDC connector with mixture of most wanted IOs and slow analog signals.

gkasprow, 2018-11-16:
We will fit 2 MMCX easily

hartytp, 2018-11-16:
Cool. Unless anyone objects, let's go for:
- 4 x SMA (2 analog in, 2 analog out)
- 2 MMC (2 digital in)
- Ethernet
- micro USB
- 4 LEDs per channel (8 total). 2 LEDs for input termination, 1 LED for loop railed, 1 LED for loop enabled and one spare

dtcallcock, 2018-11-16:
>  Or simply use IDC-BNC but we make IDC connector with mixture of most wanted IOs and slow analog signals.

Can't some of the AINs be configured as DIO to make the 'analog header' a mixture if required (assuming RC filter is fastish or easily depopulatable)?

> BNC breakout board (we can make a SMA alternative if people prefer).

Yes, now that BNC_IDC is starting to be used for other boards, an SMA_IDC would be a nice addition.

dtcallcock, 2018-11-16:
4 LEDs per channel...one spare

What colour should it be though?  I think a purple one would look cool.

hartytp, 2018-11-16:
> Yes, now that BNC_IDC is starting to be used for other boards, an SMA_IDC would be a nice addition.

:+1:

gkasprow, 2018-11-16:
Do you mean SMA_IDC in 4HP form factor like SMA_DIO ?

dtcallcock, 2018-11-16:
Yep.  

Though I do recall @jordens thinking that having used SMA_DIO, it was too dense.  Perhaps worths a bit of pain to keep it 4HP?  Hard for me to say without having used it yet.

gkasprow, 2018-11-16:
We can distribute them more evenly across the panel to make the access easier.

gkasprow, 2018-11-17:
full support for upstream and downstream EEM requires bidir LVDS transceivers that are quite costly - 12$ per set. And we don't really know at the moment how to assign peripherals.
For 3.6$  we can have ICE40HX1K-TQ144  what has 12 LVDS transceivers and enough pins to route all not used CPU pins. The bit stream can be loaded from STM32 directly, there are a few open source [projects ](https://github.com/mystorm-org/BlackIce-II/wiki/Connections-between-the-STM32-and-the-iCE40)that uses that scenario. So no FLASH or programming header is necessary. But can add for debugging purposes and keep them DNP later.

hartytp, 2018-11-17:
Thanks @gkasprow. So...the conclusion is that a downstream EEM is doable, but definitely not "free" -- once we talk about having an FPGA on this card (which is supposed to be a simple microcontroller board) we have to ask whether this is really necessary.

My feeling is:
- We definitely should have the "upstream" EEM connector to allow Stabilizer to be controller in real time by Kasli/humpback/metlino/etc
- NB since we won't clock the MCU from the EEM connector, the real time control will be asynchronous with the sample clock, so there will be a bit of timing jitter in the real time control. That's fine IMHO.
- For the upstream EEM connector, fixed directions of all the pins is fine IMHO. I'd go for:
  - pins 0-3 4-wire SPI with one CS line, using our standard pinout;
  - pins 4, 5: upstream (carrier->stabilizer) GPIO  
  - pins 6, 7: downstream (stabilizer->carrier) GPIO
- I think we shouldn't have a downstream EEM at all. If we **really** need it, let's make it a completely separate connector rather than using bi-dir LVDS drivers. Same pinout as the upstream connector, but all directions reversed.

Recapping my reasoning behind not wanting the downstream EEM functionality:
- I don't see any benefit in turning Stabiliser into a carrier for real-time control of EEMs. We already have humpback, Kasli, Metlino/Sayma for that. Why add another carrier? Making this work on a software level would require writing drivers for all the EEMs and embedding them into the stabilizer firmware. This seems like a lot of work when one could just use one of the other carriers instead
- For non real time EEMs like pounder, ethernet seems like a much better way of doing the control than using stabilizer as a carrier.
- For simple custom AFE cards like the current sense card for magnetic field stabilisation, EEM is way over the top. For those cards, it makes much more sense to just use the "digital header" on Stabiliser to provide control lines. These will be LVCMOS, which is much more convenient to deal with than LVDS for these kinds of simple applications where speed and SI aren't such an issue.

So, I don't see a clear use-case for a downstream EEM connector. Unless someone can provide a clear case where downstream EEMs are needed/really helpful, I'd prefer not to introduce them.

gkasprow, 2018-11-17:
Anyway, small FPGA would be cheaper and cosume less power  (<1mA idle) than 8 LVDS translators :)

gkasprow, 2018-11-17:
But it needs support and implementation in STM code.

gkasprow, 2018-11-17:
Let's stick with low cost FIN1027 LVDS drivers.

hartytp, 2018-11-17:
> Anyway, small FPGA would be cheaper and cosume less power (<1mA idle) than 8 LVDS translators :)
> But it needs support and implementation in STM code.
> Let's stick with low cost FIN1027 LVDS drivers.

Agreed, the FPGA has some benefits, but I don't think it's worth the software complexity in this case (although, others should feel to express themselves if they disagree). Let's stick with the FIN1027.

gkasprow, 2018-11-17:
USB won't fit on the front panel. I will place vertical one close to the CPU.

gkasprow, 2018-11-17:
@hartytp I prepared initial schematic revision.
It's here:
[Stabilizer.PDF](https://github.com/sinara-hw/Stabilizer/files/2592309/Stabilizer.PDF)

hartytp, 2018-11-17:
Wow! That was fast, thanks. I'll review it and get back to you tomorrow.

hartytp, 2018-11-18:
I believe that all feedback from here is now covered by the design review issue, so closing. Please add any further comments to the design review issue.

Thanks to all who contributed.

closed 2018-11-18 by hartytp