sinara-hw/Stabilizer #2: v1.0 rc1 design review

hartytp, 2018-11-18:
Schematic at https://github.com/sinara-hw/Stabilizer/files/2592309/Stabilizer.PDF

Review of the main analog outputs (DAC):
- [ ] Please move the 10R||10uF reference filter to between the reference and the buffer, not between the buffer and the DAC. The reason for this is that the DAC's reference input minimum resistance is only 7.5k, but very code-dependent. For Vref=4.096V, Iref=550uA, so the drop across the 10R resistor is 550uA*10R=5.5mV. 1LSB=4.096/2^16=62.5uV, so that's a very significant voltage drop, which will change a lot with the DAC code and hence kill the INL/DNL. The DAC must be driven directly from a reference buffer with very low impedance (50mOhm or so). 
- [ ] the DAC data sheet recommends 10uF||100nF on the DAC Vref pin. That's going to make the reference buffer oscillate if we're not careful. What do we do about this? The LTC6655 has very good low-frequency noise, but its broadband noise is 40nV/rtHz which is too high to connect directly to the DAC. Do we choose a different reference buffer? Do we try to isolate the buffer from the capacitive load with a fb (low DC resistance)? Or do we choose a different reference and drive the DAC directly from the reference (will this require a separate reference for each DAC to keep the cross-talk acceptable?)?
- [ ] Might be better to use a 10R resistor on the VDlogic pin as well as/instead of the ferret bead to isolate the digital signal supply from the analog supply. That helps to isolate them even at the 1MHz update rate (where the ferrite doesn't do much).
- [ ] The DAC data sheet also recommends 10uF tantalum caps near the DAC for both power supplies. Not sure if this is really necessary.
- [ ] Isn't it possible to combine analog output filtering and gain into a single stage? That would allow us to eliminate an OpAmp and do the entire thing with a single dual OpAmp.
- [ ] please annotate the schematic with the output filter characteristics (number of poles and 3dB bandwidth)
- [ ] please can we increase the output filter 3dB BW to 500kHz. We should be able to run this board at 2MSPS update rate, so 330kHz is actually a bit slow (apologies, this is my fault for putting this too low in the original specification).
- [ ] the johnson noise of the 5k resistors used on the DAC outputs is actually quite large. Is there a convenient resistor ladder with slightly lower impedance that we can use? If there isn't a convenient resistor ladder with slightly lower impedance then don't worry, these values will be ok.
- [ ] please make sure to add test points on supply lines important digital signals (as we did in other projects)
- [x] power sequencing looks good: microprocessor won't drive the DAC SPI lines until after the DAC's 5V supply is on; DAC output is high impedance so won't damage the OpAmps; reference has proper power sequencing. So I don't think there will be any issues here!

Some context here:
- Our group will likely need Fastino at some point in the next year or so
- I'd like to use exactly the same DAC and AFE for Fastino that we use here (without any modification) if possible so that we can kill two birds with one stone
- For Fastino, we will need a lot of channels so space, cost, power consumption, cross-talk are all quite important. That's why I want to minimise the number of OpAmps and find a decent way of doing the reference buffering

hartytp, 2018-11-18:
NB I'm not reviewing the power supplies section very carefully since this is copied from other projects which have already been through several design reviews. From a brief look, it looks good though!

hartytp, 2018-11-18:
Nit pick on the PSU page: in the sequencing annotation, it should be "reference" or "ref" not "adc ref" since it's used for the adc and dac.

hartytp, 2018-11-18:
- If we removed one MMCX from the FP would there be room for the micro USB? I suspect that most users would prefer USB on the FP to the 2xMMCX.
- Have you checked that there is enough clearance between the SMAs and LEDs/MMCX connectors, even when the isolating washers are mounted?

hartytp, 2018-11-18:
- [ ] The RC components on the first page are missing values
- [ ] once everything else is finished, let's have a rough power budget (useful for specifying an appropriate PoE switch to drive several boards).
- [ ] IC24 missing a PN on the schematic
- It would be good to get feedback from other potential users (@cjbe @dtcallcock @dhslichter @jordens et al) about the pin allocation on the digital/EEM connectors. Is everyone happy with the peripherals those are connected to?

hartytp, 2018-11-18:
@gkasprow I'm not going to check over the microprocessor/ethernet/USB parts of the schematic. I'm not the right person to review that.

gkasprow, 2018-11-18:
We can use another edge hang  (in-line) type of USB connector and fit in under Ethernet jack.
[image]
[image]

hartytp, 2018-11-18:
ADC:
- [ ] Like the DAC, the Iref is too large to use with a 10R filter resistor. On top of that, the capacitance could destabilise an OpAmp if there is no isolation resistor. We need to find a better reference buffer solution for this board.
- Are the LVCMOS buffers for the ADC just there to convert the 3V3 microprocessor levels to 2V5? (I'm assuming these have 3V3 tolerant inputs even when powered from 2V5?). 
- NB the ADC filter we have is quite a bit more aggressive than the one on the ADC data sheet. IIRC the bandwidth from this circuit is still fine, so that's fine IMHO (gives better rejection of high-frequency noise).

hartytp, 2018-11-18:
Also, note that the ADCs draw quite a high reference current, but there is no buffer to isolate them. That could lead to some channel to channel and ADC to DAC cross-talk. Not sure if that's an issue, but worth a bit of thought.

gkasprow, 2018-11-18:
The LVCMOS buffers are mainly to cut the noise that is propagated from CPU over the digital lines and couple with analog signals. If you want true 16bit performance it's a must.
Level conversion is another story.

hartytp, 2018-11-18:
The 4.096V reference should usually be bypassed with 10uF, rather than 2uF (in a sensible case size to ensure that the capacitance doesn't drop when the bias voltage is applied).

hartytp, 2018-11-18:
> The LVCMOS buffers are mainly to cut the noise that is propagated from CPU over the digital lines and couple with analog signals. If you want true 16bit performance it's a must.
Level conversion is another story.

Aah, good point.

hartytp, 2018-11-18:
- [ ] axuiliary ADC and DAC filters are a bit aggressive. Let's go for a 1MHz pole frequency on both (users can add a more aggressive filter on the IDC to BNC board if they wish).
- [ ] please annotate the pole frequency on the auxiliary ADC and DAC schematic page

hartytp, 2018-11-18:
> We can use another edge hang (in-line) type of USB connector and fit in under Ethernet jack.

If you think that will be okay mechanically then it's fine by me. i.e. will that be okay with the FP? will it be reasonably mechanically robust? will it be a real pain to insert the USB and ethernet at the same time? If all that's okay then it seems like a very good solution.

hartytp, 2018-11-18:
@gkasprow that's my review complete. Thanks for a really good initial version.

The only big-is thing here is deciding what we want to do about the reference buffer. Other than that it's just a few small things...

hartytp, 2018-11-18:
> The LVCMOS buffers are mainly to cut the noise that is propagated from CPU over the digital lines and couple with analog signals. If you want true 16bit performance it's a must.

If they're essential for the ADC, why aren't they essential for the DACs?

gkasprow, 2018-11-18:
ADC is more sensitive to noise due to SAR architecture. Small interference can  take a few LSBs easily.
DACs are just current sources or resistor arrays so the noise does not cause such conversion errors. The input noise can only propagate to the output. I had such issues only with ADCs so far. It is caused by ground bouncing of the silicon chip.

hartytp, 2018-11-18:
ok, makes sense. I didn't know that.

hartytp, 2018-11-18:
One other thing:
- [ ] will there be an issue with using the same 2V reference for the microprocessor that we use for the ADC? It's okay to share that between the ADC channels, since it drives a high-impedance input. If the microprocessor draws quite a bit of current (and is quite noisy) then won't there be nasty cross-talk? Wouldn't it be better to either add a really cheap 2V5 reference right by the microprocessor, or at least add a reference buffer to isolate the microprocessor?

gkasprow, 2018-11-18:
Roughly 15 years ago I built scientific grade [CCD camera](http://www.fuw.edu.pl/~gkasprow/usb_&_gbe_cameras.htm) for my master thesis. I used 16bit 10MS/s ADC. First prototype that used DRAM chips was working fine, but had higher noise (~40e-). Second one (with SDRAM, PCI and GB Ethernet) was much better with readout noise below 10e-. And to my frustration, every image had same pattern applied. It was roughly 10LSB pattern, same on every image. I tried many things - separated ADC from FPGA board, supplied with batteries, placed in metal cage. After hours of fight I noticed that only digital lines and GND are connecting the two. So I placed HCT244 buffer. The pattern disappeared and overall noise was greatly reduced to just 2 LSBs RMS including front end. It was even smaller than in datasheet. Their reference design did not use such buffering :D AFAIR I used AD9826 chip. The pattern was caused by SDRAM controller that went through exactly same sequence during every image acquisition that caused power supply bouncing. This signal went through digital lines to the ADC and coupled with analogue lines via internal capacitances and also impedance of ground  bondings.

gkasprow, 2018-11-18:
Since that time I always place buffer between ADC and readout IC :)

hartytp, 2018-11-18:
ouch.

hartytp, 2018-11-18:
One other question: does one have to place a jumper to flash new firmware? If so, please can we add an annotation about that. Is there any way around that to make loading new firmware remotely (without having to add then remove a jumper) possible?

gkasprow, 2018-11-18:
Firmware update can be triggered by software command.

hartytp, 2018-11-18:
okay, good.

hartytp, 2018-11-18:
One other slight nit pick:
- [ ] let's call them DI0 and DI1, not "adc trigger" and "dac trigger" (they can be used for different things in fw so let's keep the names general)

hartytp, 2018-11-18:
@gkasprow so, the plan for the GPIO/DIO header is that we have 16 single-ended signals. If this is used with the IDC to BNC board to break them out then we must set 8 of those signals to LOW in firmware, and use the other 8 as GPIO. Are you happy that this plan will work okay?

gkasprow, 2018-11-18:
@hartytp I' not sure if grounding BNC with CPU pin is good idea. It will damage CPU quickly with first serious EMI from i.e. nearby thunderstorm.

hartytp, 2018-11-18:
@gkasprow ack. That's why I asked.

So, with the current arrangement the digital header is not compatible with the IDC BNC board, because that assumes that there are 8 differential signals on each header.

I can see a few options options (if you have a better solution let me know!):
1. Create a new revision of IDC_BNC that has jumpers that allow one to switch the BNC grounding between circuit ground and the _N half of the differential pair
2. Split the GPIO across two digital headers, each of which only has 8 signals.
3. If there are enough pins, then we can have two digital headers. One header has 8 single-ended signals and is designed to be compatible with the IDC_BNC board. No need for SPI/I2C or anything fancy here. Stick with GPIO and maybe a timer output. The second header has more digital signals with a mixture of GPIO/SPI/I2C etc. The second header is designed for connecting to custom AFE modules and is not compatible with the IDC_BNC board (it should use a different header so they cannot be mixed up by mistake). We should have clear annotations on the schematic labelling which headers are intended to be used with IDC_BNC.

If there is room then 3 sounds like the best option to me. What do you think?

hartytp, 2018-11-18:
- [ ] Let's also make sure to expose the main 12V supply on one of the board-board headers (probably on the high-density digital one) using a few pins to give 3A or so. The use case I have in mind for this would be putting a 5A SMPS on an AFE mezzanine to create a compact ultra-low noise current source for coils etc.

hartytp, 2018-11-19:
@gkasprow what do you think we should do about the reference buffers?

Looking at the LTC6655 data sheet, the voltage reference regulation starts to degrade after 1mA. Each DAC draws up to 0.55mA, and the two ADCs draw something comparable. By 2mA, the error due to the reference alone is something like 20ppm, so we will see a slight degradation of INL/DNL and cross-talk.

For Fastino with >=8 DACs per board, we definitely cannot drive the DACs directly from the reference. So, finding a good way of doing the reference buffering is going to be important. What's your preferred solution to this? Do we need to pick an OpAmp and design an RC compensation network to ensure both stability and low output impedance at all frequencies? The latter is doable, but will take some time in spice (unless you already have a working design for this?)

dtcallcock, 2018-11-19:
> finding a good way of doing the reference buffering is going to be important. 

What about the circuit examples in the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/6655ff.pdf)?

hartytp, 2018-11-19:
The LT3042 is a lovely LDO, which I've used as a ref buffer before. Load regulation is 0.5mV (max) for a change in load between 1mA and 200mA. Assume that's roughly linear (no idea how valid that is) and it gives approx 2.5mOhm output impedance.

Assume peak-peak change in ref current is 500uA*num_comverters. This gives 1uV/converter. For 4.096V reference, 1LSB=62uV, so that should allow us to drive up to 62 converters from a single reference buffer while still keeping the worst-case cross-talk below 1LSB pk-pk.

However, that's only at DC. Once we start adding 10uF per converter, what happens to the ref buffer BW? The concern would be if a frequency band appears where the reference is running out of bandwidth (and hence its output impedance rises), but the decoupling capacitors aren't low enough impedance yet (the DACs require something like 50mOhm max). That's not something that can be guaranteed from the specifications, and would need testing. That's if the thing is even stable with that much output capacitance, which isn't clear to me from the data sheet. For a few DACs and ADCs it's probably fine, but it would need testing. This path doesn't seem crazy to me and might be the best option for this design.

We could obviously use one LT3042 per converter, but that would be very expensive and bulky.

As to the other circuits on the LTC6655 data sheet (essentially just an emitter-follower), they're less likely to work. One would need a noise filter to reduce the 40nV/rtHz reference noise, while still maintaining enough BW to keep the output impedance low at higher frequencies. That would probably require an OpAmp circuit to drive the BJT, which would require careful design.

The other option is to pick a cheap OpAmp in a small package and design a compensation network to make it work. Make it cheap, compact and low enough power that we can use a separate buffer per channel. That has the advantage of providing excellent isolation between the channels.

hartytp, 2018-11-19:
My guess is that @gkasprow has done this many times before and already has a preferred solution.

hartytp, 2018-11-21:
> How about using MMCX?

If you're going to use coax then might as well just use the FP SMAs IMO. The internal MMCXs are okay, but can be a bit of a pain to route with all those IDCs.

My primary use-case for the "main analog" header is for plugging directly into custom AFEs like a current stabilization circuit, which screw directly to stabilizer and use the headers for board to board connections. That seems much nicer than using coax for board to board stuff.

dtcallcock, 2018-11-21:
> My primary use-case for the "main analog" header is for plugging directly into custom AFEs like a current stabilization circuit, which screw directly to stabilizer and use the headers for board to board connections. 

Ah, I had forgot that you wanted to do that.

hartytp, 2018-11-24:
@gkasprow NB when you do the routing: if we connect more than one converter to the same ref buffer then we must make sure to use a start layout for the reference connections to minimise the cross-talk due to trace impedance.

gkasprow, 2018-11-24:
The original circuit that converts unipolar to bipolar output relies on internal DAC resistors.
We can combine gain stage with shifter circuit but have to connect external reference voltage to not include precision resistors in each channel. So we can generate 4/3 *4.096V reference and use the circuit below:
[image]

gkasprow, 2018-11-24:
edit: reference value

gkasprow, 2018-11-24:
3.413V is 4.096 * 5/6

gkasprow, 2018-11-24:
This can be achieved using 5 + 5 + 10 + 10 resistor, resistor ratio of 5:25 gives  1/6 and 5/6 which does the job.

gkasprow, 2018-11-24:
Buffering references is challenging. I can drive 10uF with 30Ohm series resistor and still maintain DC accuracy, but it requires additional feedback loop (1 resistor + 1 capacitor).
Another idea is to use op-amp with external compensation that can drive indefinite capacitance, but I don't know any with very low offset (they typically have a few mV of offset)
Yet another idea is to use precise opamp in every DAC channel as we do for Stabilizer.
For Fastino it would increase the cost too much. We can also try to isolate the opamp with LC filter, but it is tricky.

gkasprow, 2018-11-24:
The question is how big the isolation resistor can be. AD5542 defines in fig 17 that current changes between 100uA up to nearly 200uA. So our buffer must keep ref stability below 1 LSB, in this case 62uV 4.096/2^16). Iref change is 100uA, so maximal Rs = 0.62Ohm. This is also the maximum value of reference buffer. If we want to drive 32 channels from same reference, our buffer must withstand max and min load for all buffers. So its output resistance must be lower than 0.62Ohm/32 =  19mOhm. Since the DACs change the reference current rapidly, the buffer must react quickly and not oscillate. 
The LT3042 you proposed could work, but with tantalum capacitors which have much higher capacity.
Analog Devices in [CN0079 note](https://www.analog.com/media/en/technical-documentation/application-notes/an-1525.pdf) presents idea of using individual AD8628 with feedback taken via REFS.
Such solution would increase the channel cost by roughly 2 EUR. On the other hand,  cost of AD5542 (10EUR), AD8620 (11EUR) and AORN2-1AT3 (2.7EUR) is already 24EUR so additional reference buffer is less than 10%. AD8628 is available in TSOT so we can fit them easily and use  it for distribution of other references due to very low offset (1uV)

gkasprow, 2018-11-24:
AD8628 in SOT-23 costs 1.5EUR@100pcs

gkasprow, 2018-11-24:
Another issue can be with delivery of 3.413V reference. But here situation is easier. Every channel consumes +68uA / -340uA of current, but the impedance is always 10k. Even with 32 channels we still have 312Ohm.
The reference buffer must maintain -2.1mA / +10.88mA but every opamp can do that. Of course at high frequency its performance will degrade and the crosstalk may get worse. We must make sure that its BW is much higher than amplifier BW.

hartytp, 2018-11-25:
Regarding the reference buffer, I can see a few options:
1. LT3042. Maybe limit to max 8 converters per reference buffer to limit cross-talk. For Fastino we'd need quite a few reference buffers. The DC output impedance of this buffer is extremely low, and it has a wide bandwidth when loaded with 4.7uF, allowing it to keep a low impedance across a broad frequency range. The biggest issue I see here is how well the reference buffer performs at high frequency when we load it with a lot of capacitance. However, since the buffer is quite fast, the best bet might be to place it physically near the converters, connecting it with short wide traces to keep the trace impedance low. We can then use something like 1uF capacitors on each reference, so the total capacitance on the LT3042 output is not too high.  
2. Add 1 OpAmp buffer per converter. Pick a fast OpAmp with good current drive so that no capacitor is needed on the reference pin (i.e. we rely on feedback to keep the reference impedance low at all frequencies, rather than the standard approach of using capacitance). The big concern here is that while this will work well at DC, if the OpAmp isn't fast enough, this might lead to worse DAC glitches/AC non-linearity since the reference source impedance can rise at higher frequencies.
3. Use 1 OpAmp buffer per converter. Add the usual 10uF||100nF capacitors, add an isolation resistor + RC compensation network to the OpAmp to stabilize it. This would need careful simulation.

The challenge with any of these approaches is ensuring that the reference pins have see a low impedance across all frequencies.

@gkasprow which solution do you think we should go for. I'm tempted by (1). It's definitely the cheapest and should work well so long as we watch the load capacitance on the LT3042 as well as the trace impedances. What do you think?

hartytp, 2018-11-25:
> Another issue can be with delivery of 3.413V reference. But here situation is easier. Every channel consumes +68uA / -340uA of current, but the impedance is always 10k. Even with 32 channels we still have 312Ohm.

It's not really easier here. While there is a 10k resistor, the effective impedance changes significantly (see how much the current changes by). We still have a large, non-static current draw on this reference, so it's a hard problem to solve.

As always, it's easy at DC, but avoiding glitches/cross-talk/poor AC INL is the tricky part. Much better to avoid references driving summing junctions where possible

gkasprow, 2018-11-25:
That's why I wrote that we need fast opamp. Glithes are not an issue here because are filtered by first stage opamp. Let's try with combined filter and gain stage.

hartytp, 2018-11-25:
> Let's try with combined filter and gain stage.

Sounds like a good plan. Can you do a quick spice sim of that?

hartytp, 2018-11-25:
> Can you do a quick spice sim of that?

Otherwise, if you post a schematic then I can do a SPICE work up some time this coming week.

gkasprow, 2018-11-25:
this could work
[image]

quick simulation of second stage

[image]

gkasprow, 2018-11-25:
The only weak point is output resistance of the DAC that has 20% tolerance. I added 10k to reduce its influence, and made the first stage slightly above 500kHz. It's worth to simulate how it effects the BW and also performance with real model of AD8620.

gkasprow, 2018-11-25:
Simulation with ADIsimPE
[image]

gkasprow, 2018-11-25:
This 300pF capacitor looks surprisingly large. so there is sth wrong with design tool I used.

gkasprow, 2018-11-25:
I found where the problem lies.
The AD8620 has 15pf of input capacitance. And filter wizards do not take it into account. The DAC output resistance with input capacitance already makes 500kHz LP filter. So when I added 10k resistor to mitigate 20% tolerance of internal resistance, I cut the BW to 377kHz.

gkasprow, 2018-11-25:
Edit - I used wrong resistances at the DAC level shifter.
When I change them from 10k do 28k as DAC has, opamp input capacitance increases the BW up to 700kHz! So we have to add compensation capacitance in the first stage as well
[image]

gkasprow, 2018-11-25:
In this way we achieve:
[image]

gkasprow, 2018-11-25:
Let's analyse the first stage only. Without compensation we get:
[image]

The input capacitance on negative terminal of opamp is responsible for such peaking.

gkasprow, 2018-11-25:
With 15pF compensation capacitor we get smooth response at 1.9MHz. So let's add some resistance to the DAC output and also some capacitance to the opamp input to make the filter less sensitive to the DAC output resistance variation. In semiconductor process the capacitances are controlled with high accuracy, but it's hard to control resistances. But one can control resistance proportions, that's why the Rinv and Rfb have 20% tollerance but very good matchning.
So let's add 10k and 5p to get 500kHz first order filter.
[image]

gkasprow, 2018-11-25:
Of course when we combine 2 filters, we will get 500kHz cut-off at -6dB so individual BW must be trimmed.
And have to compensate for op-amp capacitances otherwise will get peaking above 1MHz.
[image]
[image]

hartytp, 2018-11-25:
@gkasprow thanks for doing that! That response curve looks really nice. The plot with component tolerances taken into account looks good too. I'm happy with that design for the filter. Good work.

[87 more replies]

closed 2018-11-30 by hartytp