sinara-hw/Stabilizer #5: backplane support

gkasprow, 2019-01-02:
Shall I add support for CPCIS backplane  in this revision?

hartytp, 2019-01-02:
@gkasprow is there a finalised proposal for how the BP support will work in the EEM ecosystem, which has been signed off on by all interested parties (in particular, @jordens, who has been a key driver of this)?

I don't object to adding BP support if there is a concrete proposal which everyone is happy with, and if it won't cause delays/other problems, but it's also not something that I see as particularly important right now. Others may disagree.

gkasprow, 2019-01-02:
Backplane support is optional and CPCIS connector won't be mounted by default.

hartytp, 2019-01-02:
@gkasprow sounds good then. Are there some notes on this somewhere (e.g. the Sinara Wiki) with an explanation of the backplane/connector connectivity and how we plan to use it in Artiq? Also, even if we assume that we fix all modules to be 4HP by swapping BNCs for SMAs/MCXs, how will we handle the case of cards that require 2EEM connectors?

jordens, 2019-01-02:
2 EEMs per board and 8 HP options are all possible and discussed in https://github.com/sinara-hw/meta/issues/30
Yes. A summary of the consensus there should be written up.

gkasprow, 2019-01-02:
The discussion was [here](https://github.com/sinara-hw/meta/issues/30)
2EEMs will be supported by additional P6 connector that has second EEM signals routed.
Some slots would be capable of supporting 16LVDS channels. There are several options including lane switching in the controller but they don't interfere with module specification nor backplane.

hartytp, 2019-01-02:
Okay. If you're happy with the BP design and think it will be useful for people then I have no objections to including it.

closed 2019-01-11 by gkasprow