sinara-hw/Stabilizer #6: Use FT2232 USB UART+JTAG

cjbe, 2019-01-05:
@gkasprow @hartytp I think we should use a FT2232 for the USB interface, rather than using the STM32 onboard USB phy. This means we don't have to implement software support for USB. It would also allow us full JTAG access to reprogram the device (rather than having to use DFU), and allow easy debugging access.

Changes required:
FT2232 ch 0:
0 -> JTCK
1 -> JTDI
2 -> JTDO
3 -> JTMS
4 -> NRST (system reset)

FT2232 ch 1: connect to USART (PA9, PA10)

gkasprow, 2019-01-05:
USB UART with STM32 is trivial, just a few lines of code. Simply use CubeMX, enable USB support and use example design. I did it many times during classes with students.
 I'm not sure if popular STM32 tools use FTDI  as JTAG debugger. 
@jordens do you confirm that FTDI JTAG would work as STM32 debugger?

cjbe, 2019-01-05:
@gkasprow I think the plan is to use Rust for the Stabilizer firmware. There is a Rust USB stack in active development that could be made to work, but would require writing a USB phy wrapper.

The STM32 JTAG just exposes a standard Cortex M7 DAP, and is supported by openocd (e.g. https://github.com/ntfreak/openocd/blob/master/tcl/target/stm32h7x.cfg). This means we can use the same tooling and debug workflow as we do for the standard Artiq stack.

cjbe, 2019-01-05:
@jordens the SWD connections are a subset of the JTAG connections, and openocd implements SWD. This means that with the proposed connections we should also be able to do SWD.

The subtext to all of this is that implementing and debugging a USB stack in Rust sounds like a time consuming (=expensive) detour, but I also don't fancy doing this in C with the ST libraries. This FT2232 option gives us a foolproof UART for debugging, and allows robust in-situ reprogramming that will work easily (=cheaply).

jordens, 2019-01-07:
Yes. I know openocd and SWD. It also gives you really nice and fast debugging and console output that would be useful for data transfers. Probably much easier and faster in the initial debugging phase than the UART or the network.
I would still recommend not putting the FTDI stuff (FTDI, EEPROM, crystal, JTAG muxing) on there and leave the USB bus flexible and open for DFU, data, HID use. But your choice.