sinara-hw/Stabilizer #10: Stabilizer errata

gkasprow, 2019-03-05:
- [ ] ceramic capacitor below J12 connector

gkasprow, 2019-03-07:
- [ ] missing connection of 2V reference port with reference label. Short pin 6 RN3C with pin 3 RN4C

jmilczarek, 2019-03-13:
- [ ] micro USB port placed too far beyond PCB outline - creates tension of the front panel

gkasprow, 2019-03-13:
@jmilczarek the connector was placed on purpose to protrude via the panel, otherwise one won't be able to fully plug the USB plug. So one needs to enlarge the panel hole. Please post the image from front side. You can drag and drop the image to the github.

gkasprow, 2019-03-26:
- [ ] add RN designators on silkscreen

dtcallcock, 2019-05-31:
C164 should be 82pF, not 390pF.

gkasprow, 2019-06-03:
- [ ] 3V3MP LED (LD11) annotated as P3V3

hartytp, 2019-06-03:
> 3V3MP LED (LD11) annotated as P3V3

Is this something that's just on the EEM connector? Should it be annotated something like 3V3_EEM? One thing I found confusing when I first powered the board on is that the 3V3 light doesn't come on (usually, one LED in the strip of power on lights not coming on is a very bad sign with these designs, but here it's normal).

gkasprow, 2019-06-04:
True, MP means management power which is also confusing.

jordens, 2019-06-04:
Please label it (P)3V3(_)MP.

* We have that label on many other places on many other EEMs. Changing it is inconsistent and adds confusion.
* 3V3_MP being off hasn't been a problem in other EEMs.
* 3V3_EEM does not indicate that it can be off anymore than 3V3_MP does. It would not help.
* I don't see the problem with the label. Management power has clear meaning and I can't come up with something clearer (EEM certainly isn't clearer).
* But it would be good if you add that alternative supply path. It enables (a) standalone ops with the EEPROM and (b) enables driving another EEM (modulo pinout).

gkasprow, 2019-06-04:
OK. The last request is already on top of this issue.

airwoodix, 2019-06-05:
- [ ] fix front panel or schematic LEDs labeling (rather the latter)
On my v1.0 board + FP, I have the following:
  - FP_LED0 (green, PD5) <-> L2 on FP
  - FP_LED1 (red, PD6) <-> L3 on FP
  - FP_LED2 (green, PG4) <-> L0 on FP
  - FP_LED3 (red, PD12) <-> L1 on FP

closed 2019-08-19 by gkasprow