sinara-hw/Stabilizer #9: Stabilizer tests

gkasprow, 2019-03-05:
The Stabilizer board arrived!
[image]

hartytp, 2019-03-06:
Nice! When can you ship us one to play with?

hartytp, 2019-03-06:
Looks great.

gkasprow, 2019-03-06:
@hartytp let me make sure all critical parts work.
So far power supplies work.
POE works
I want to check if CPU can be programmed and if AFE works, then I will send it to you.

gkasprow, 2019-03-06:
STM32 works in DFU mode

gkasprow, 2019-03-06:
JTAG works as well. I managed to compile and burn simple init programe using Atollic IDE

gkasprow, 2019-03-07:
I found one issue with ADC AFE, fixed it and now it works fine. I will send to @hartytp the boards tomorrow.

gkasprow, 2019-03-07:
Ethernet PHY also works.

hartytp, 2019-03-07:
Whoo! Nice, thanks Greg.

hartytp, 2019-03-07:
Out of curiosity, does it come with a panel? I didn't get panels for my Thermostat boards yet...

gkasprow, 2019-03-07:
yes, they are preparing panels.
For Thermostat there is panel option or stand-alone enclosure option. Which one do you want?

hartytp, 2019-03-07:
What's the standalone option? For now I'd probably take the panel and then put it in a standard euocard mini-rack using off-the-shelf FPs to break out the IDC to D-types.

gkasprow, 2019-03-07:
The board fits handy aluminum Hammon enclosure.

gkasprow, 2019-03-07:
Look [here](https://raw.githubusercontent.com/sinara-hw/Thermostat/master/DOC/Thermostat%20standalone_2.PNG)

hartytp, 2019-03-07:
Nice! Can you add that PN to the Wiki please (don't think I have write access).

Let's finish the conversation off by email though.

hartytp, 2019-06-04:
cf https://github.com/quartiq/stabilizer/issues/9

hartytp, 2019-06-04:
@gkasprow before we commit to using this PoE module in too many designs, we should look into the noise properly and check it's actually up to our spec (and, e.g. that we don't need more screening/filtering/CMCs/etc)

gkasprow, 2019-06-04:
This PoE module has already LC filter at its output.
But it may cause noise injection between input and output. We may need to add CMC at the input and some cap between input GND and output GND to bypass the noise source.
How can I recreate the measurement in my lab?

hartytp, 2019-06-05:
@jordens can you post the code you used to take that data? If not I'll hack something up.

jordens, 2019-06-05:
I don't have it anymore. It was a quick and early test to check data integrity and network stack fundamentals. Just buffer and packetize the iir cycle number and all the four channels (2*(1+4) bytes at 500 kHz are just 40 Mb/s).

gkasprow, 2019-06-05:
Do you have the binary I could burn in FLASH and try?

hartytp, 2019-06-05:
@jordens am I right in thinking that you also saw noise on the DAC output due to the PoE? Was that visible with the DAC in its power-on default state? If so, that's probably the easiest way for @gkasprow to debug this.

hartytp, 2019-06-05:
@gkasprow I've loaded @jordens firmware from https://github.com/quartiq/stabilizer

I hacked the firmware to set dac0 to 0V (0xffff >> 1) and dac1 to max (0xffff). Looking at both channels on a scope.

With a 12V supply (no PoE) I see 

[image]

hartytp, 2019-06-05:
Same but with a PoE switch instead of the 12V supply

[image]

hartytp, 2019-06-05:
@gkasprow there is some ~150kHz noise on both DAC channels:
- it's not due to ADC/DAC sampling (sample rate is higher, and it's unchanged if I pause the ADC + DAC updates)
- seems to scale with the output voltage (checked 0V, +5V and +10V outputs).

So, looks like SMPS noise on the reference. Any suggestions as to the culprit @gkasprow , or should I dig around with a scope probe?

gkasprow, 2019-06-05:
Did you try to see what you measure on the crocodile GND clip when it is attached to the Sampler GND?
In this way, you can observe the noise induced in the scope ground loop.

hartytp, 2019-06-05:
> Did you try to see what you measure on the crocodile GND clip when it is attached to the Sampler GND?
> In this way, you can observe the noise induced in the scope ground loop.

(a) I'm using a SMA from Stabilizer to the scope rather than crocs
(b) if it were a ground issue then the noise wouldn't scale with the DAC output voltage

hartytp, 2019-06-05:
There definitely is some noise due to the PoE converter. Prob not really surprising, but resting a finger on the power inductor's case gives me a huge amount of noise on the outputs

jordens, 2019-06-05:
Try bypassing the output CMC. I vaguely remember that that made a difference.

hartytp, 2019-06-05:
@jordens it's not obvious to me how that would explain the issue, given that the noise scales with the DAC output. Since I have no load current, I'd expect the CMC pickup to be linear.

jordens, 2019-06-05:
And (obviously) keep the board underside away from things like an antistatic mat.

hartytp, 2019-06-05:
> And (obviously) keep the board underside away from things like an antistatic mat.

And try not to spill too much tea on it while taking measurements etc

hartytp, 2019-06-05:
@gkasprow I wasn't able to find an obvious source of this noise. It's also a little tricky since there are quite a few noise sources on the PCB so whenever I probe any node the pickup is so high I can't see the signal I'm looking for. 

Do you think you can have a look at this? Stick the DAC output onto a scope via an SMA. So long as it doesn't power on to 0V you should be able to see this.

hartytp, 2019-06-05:
I measure 1.8LSB RMS for both G=1 and G=10 (same as stabilizer), so no evidence of anything untoward there. Good!

@gkasprow do you think the DAC output filter is slightly unstable? Otherwise I'm surprised to see so much noise on the DAC and not on the ADC.

gkasprow, 2019-06-05:
Sure, I will have a look at this. I don't have any stabilizer right now.
I will do it at home, which is ~20km from Warsaw centre where I don't have interferences which make any low noise measurements at WUT difficult.

hartytp, 2019-06-05:
Thanks!

AFAICT Stabilizer is the biggest noise source (plenty of SMPSs etc) so no need to worry too much about that...

hartytp, 2019-06-06:
@gkasprow I've hacked @jordens' firmware to dump samples at max rate over ethernet. Let me know if you want it and I'll give it to you, otherwise I'll tidy before posting (I also have some vague plans to combine this with PRNG noise added to the DACs for FFT-based transfer function measurements).

Anyway, here is the data I see with:
- ADC0
- input disconnected, but termination on
- driven from a PoE switch (netgear JGS516PGE)
- sample rate is approx 500kHz
- PGIA gain is x10

I see 1.4 LSB RMS noise with a basically white spectrum. I don't see any evidence of any particular noise spurs...Good!
[image]

hartytp, 2019-06-06:
There are some spurs at around a few kHz, which may be aliassed down from higher frequencies and a small spur around 190kHz.

hartytp, 2019-06-06:
Same but PGIA gain 1, not much change...

So, all in all, this looks really good apart from the noise on the DACs. Good job @gkasprow Thanks also to @jordens for the lovely firmware.

hartytp, 2019-06-06:
[image]

hartytp, 2019-06-06:
Same but with some filtering and log log axes. I suspect the few kHz spurs are an aliassed version of the spurs @jordens saw here https://github.com/quartiq/stabilizer/issues/9
[image]

jordens, 2019-06-06:
At a PGIO gain of ten, I expect a LSB of 30µV, not the 300mV shown in https://github.com/sinara-hw/Stabilizer/issues/9#issuecomment-499465401 . Could you double check that?

hartytp, 2019-06-06:
@jordens yes, sorry, I forgot to rescale the voltage plots after changing the PGIA gain.

The 1.8LSB RMS number is correct for both gains, but the voltage conversions in the plots I posted are only correct for G=1...

hartytp, 2019-06-10:
@gkasprow the code I used for these tests is here: https://github.com/OxfordIonTrapGroup/stabilizer/tree/adc_log

When you connect to port 1234 it will stream raw samples at you.

hartytp, 2019-06-10:
@gkasprow let me know if you want me to send you a stabilizer card to look at the DAC noise issue...

gkasprow, 2019-06-10:
Technosystem has one piece that they can lend me.

hartytp, 2019-06-26:
Step response looks good. 600ns rise/ fall
[image]
[image]
[image]

hartytp, 2019-06-26:
Given that much of the design is well tested from other designs like Sampler I’m now happy that all works well apart from the dac noise issue. Once that’s solved let’s push out a new release.

hartytp, 2019-07-12:
@gkasprow do you have a spice model for stabilizer dac afe? I got the impression that it’s maybe a factor of a couple higher noise than I expected would be good to compare to the model.

gkasprow, 2019-07-12:
It's [here](https://www.dropbox.com/sh/iv3qgrgfrsztj9w/AAB7AWFddpc1Nh-4yLUgz0Xma?dl=0)

closed 2019-09-06 by hartytp