sinara-hw/Stabilizer #12: Adc spi buffers

jordens, 2019-03-19:
The buffers have 4.3 ns prop delay. In roundtrip that adds up twice to the adc clk to sdo delay and the stm32 miso setup delay. This limits sample rate and transfer speed. We could play some tricks with 17 bit and cpha=0 but it's not nice. Reconsider this at some suitable time.

gkasprow, 2019-03-19:
you can enable some delay to other lines using CPLD :D But it is not elegant.

jordens, 2019-03-20:
Adding more delay somewhere else doesn't solve the issue.

gkasprow, 2019-03-20:
What can we do here? Change the CPU to something else? Or just changing the SPI assignment solves the issue? 17bit SPI can make issues with DMA.

jordens, 2019-03-21:
We first have to decide whether those buffers are really required. If not, then remove them. If yes, then maybe there are faster ones. If not, then we can use falling edge sampling (a bit tricky, 17 bits, discard first, cpol=0, cpha=1, would work better with the 32 bit wide SPI123 buses). Or we live with it and tweak the clocking so that the ADCs run at >34ns (which would also require them to be both on SPI123 or both on SPI45 for clocking reasons and it would preclude the merging of a DAC/ADC pair on a single SPI bus).
DMA (of the SPI data) may not even be useful/required for the dominant application and I don't see an issue with it for 17 bits right now.

hartytp, 2019-03-21:
> We first have to decide whether those buffers are really required. If not, then remove them. 

@jordens yes.

The motivation for these buffers wast that @gkasprow had seen issues in the past where noise on the digital lines coupled into the ADC and degraded its noise floor. I assume that this effect is strongly dependent on the particular ADC, and it's not obvious to me whether or not it will be an issue in this design.

To know that, we'd need to test. If anyone has time to do that then it would be great. Otherwise, my assumption was that -- at least in the applications I have in mind -- we can live with the delay and just slow down the ADC. But, I haven't looked into the clocking options, so that may not work.

gkasprow, 2019-03-21:
We may use faster logics, i.e. AC logic
I observed such issues with FPGAs. Maybe this particular CPU won't inject so much noise.

jordens, 2019-03-24:
Maybe the cnv buffer if the only one that we need. Or cnv and clk. @gkasprow did you see jitter/phase noise or amplitude noise? The former is not that critical for us. Iirc it was a ccd adc. That would also be mostly sensitive to amplitude noise/injection.

gkasprow, 2019-03-24:
It was an amplitude noise. The BW was too small to observe the clock jitter influence.

hartytp, 2019-06-06:
I'll try and have a look at this at some point in the next few days.

hartytp, 2019-06-10:
@jordens @gkasprow I'm happy to test this, but first I'd like to clarify what the plan is: the ADC LVCMOS inputs run from a max 2V5 supply. Am I right in thinking that the microprocessor uses fixed 3V3 logic levels? If so, how would you plan to do the level conversion?

hartytp, 2019-06-10:
The other option is to replace them with a faster buffer like http://www.ti.com/lit/ds/symlink/sn74auc2g34.pdf (@gkasprow are you aware of any faster parts that are still footprint compatible with the ones on Stabilizer?)

At that point the propagation delay doesn't affect the overall timing budget much.

unless anyone has a better idea, I'll order some of those, swap them out, re-check the noise. If that looks good I suggest we make the change for v1.1.

hartytp, 2019-06-10:
SN74AUC2G34DCKR would be the replacement PN

gkasprow, 2019-06-10:
the 74AUP1G125GW I used accepts 3.3V at its input and protects ADC against 3.3V signals from CPU
In the other direction, 2.5V is still logic H a the STM32 input so one doesn't need a level translator

hartytp, 2019-06-12:
@gkasprow okay, so how about we scrap the ADC output buffers and replace the input buffers with SN74AUC2G34? Any objections?

gkasprow, 2019-06-12:
I'm fine with that.

jordens, 2019-07-12:
Getting rid of the buffer on MISO and speeding up the buffer on CLK  by a couple ns is all that's needed.

hartytp, 2019-07-12:
I'm planning to run a quick test with the SN74AUC2G34 I have in the next couple of days. But, it probably makes sense to use a different part with better availability for the next revision (just chose that as it was footprint compatible with the current part).

gkasprow, 2019-07-12:
you can replace them with 100r resistors

hartytp, 2019-07-19:
@gkasprow the part I quoted above was a 2x buffer, so not a valid replacement. Do you know of any faster, pin-compatible replacements for 74AUP1G125GW?

> you can replace them with 100r resistors

You mean to limit the current through the ADC protection diodes?

gkasprow, 2019-07-24:
Yes, this resistor is to protect against latch-up event.

gkasprow, 2019-09-05:
we can use dual [buffer](http://www.ti.com/lit/ds/symlink/sn74auc2g34.pdf) which has 0.8...1.6ns delay at 2.5V
Alternatively we can use [241](http://www.ti.com/lit/ds/symlink/sn74auc2g241.pdf) or [126](http://www.ti.com/lit/ds/symlink/sn74auc2g126.pdf) buffer

gkasprow, 2019-09-06:
I used SN74AUC2G34DCKR

closed 2019-09-06 by gkasprow