sinara-hw/Stabilizer #13: Spi bus assignment

jordens, 2019-03-19:
That uC has spi123 different from spi45 and again different from spi6 in terms of features, bus matrix, dma, and clocking connections. Make spi12 the adcs, spi3 the eem, spi45 the dac and spi6 the gpio header.

gkasprow, 2019-03-19:
can they be connected in better way using CPLD?

jordens, 2019-03-20:
The cpld adds yet more delay slowing down the adc spi clock and it adds several layers of compelxity. As I said I think it's 
unnecessary and suboptimal and I'd avoid it if possible. It's not a good idea to put a circuit on a board without a detailed plan when and how it is supposed to be used. Especially if this can be solved by design which is easier, cheaper, and has better performance.
I'd prefer to make seeing changes like this one as soon as possible before the next big batch is produced.

jordens, 2019-03-20:
This is not critical and it's not limiting right now. But it would be better to clean it up.

gkasprow, 2019-03-20:
The CPLD was added just in case we have issues with SPI Chip select generation in DMA mode. I had such issues in the past with F4 series CPUs. The idea is to get rid of it in the next prototype. Swapping SPI is not a problem. Actually, the firmware template should be created before completing the HW part to avoid such issues.

jordens, 2019-03-20:
Yes. so far I see no need for the cpld.
But this is not just templates. E.g. if it turns out that the spi buffers are needed and pseudo-17 bit transfers, sampling on the first edge, discarding the first bit is required to get sufficient speed, then only the first three buses would be suitable.
I would also consider merging one adc and one dac on a bus. But only after further thinking and testing this in software.

gkasprow, 2019-09-05:
@jordens can I use SPI 2,3 for DACs and SPI 4,5 for ADCs? The SPI5 MOSI conflicts with ADC IN2. So if we want to use your configuration, the AUX IO IN2 won't be differential.

jordens, 2019-09-06:
Originally, this was to tackle the SPI buffer delays (#12) with sampling trickery in the SPI peripherals that only works on SPI123. With faster buffers that's probably not be necessary anymore.

If the NSS of the ADCs are on timer output channels of the same timer type (i.e. "simple") and if the ADCs and DACs use SPIs from the same clock domains (SPI123 and SPI45 respectively), then this is OK.

gkasprow, 2019-09-06:
@jordens are you using ADC SPI in slave mode and use timer outputs to generate the NSS for both SPI and ADC?

gkasprow, 2019-09-06:
Or, you trigger ADC conversion with NSS and read in with SPI in master mode in the timer interrupt?

jordens, 2019-09-06:
Currently it's a timer that triggers a DMAMUX that drives two DMAs to the SPI configuration registers which in turn start transactions using the SPI peripheral MIDI/MSSI delays to comply with the ADC timings. On completion of the first ADC xfer an IRQ is fired that does the computation and starts the SPI transfers to the DACs. Some of it is not strictly documented (e.g. driving two DMAs from one timer or DMAing the configuration register). It might be easier and more configurable to rewire it with timer channels on NSS and separate channels to the DMA-SPI transfers (master or slave). The current way does however achieve really good latency.

dnadlinger, 2019-09-06:
For using Pounder for fibre noise cancellation, it would be useful to have a high-bandwidth link to the (presumably) [AD9959](https://www.analog.com/media/cn/technical-documentation/data-sheets/AD9959.pdf) on the AFE board. It's been a while since I've looked at the SPI clock tree/DMA structure, so I can't immediately comment on what the best bus assignment for that would be.

gkasprow, 2019-09-06:
@jordens OK, now it's clear. You want to trigger the timer with NSS signal?
If we choose SPI45 as ADC Rx master, the SPI5_NSS would be connected to TIM16_CH1, 
SPI4_NSS would be shared with TIM15_CH1N. Is it acceptable?
If not, I can additionally connect these NSS lines with some other timer pins.

gkasprow, 2019-09-06:
I've read that in some STM32 one can swap MISO with MOSI. Is it possible with SPI5?

gkasprow, 2019-09-06:
There is an option called IO swap. Does it really swap MISO with MOSI so I could use SPI5 for  DAC?

gkasprow, 2019-09-06:
It seems to [work](https://community.st.com/s/question/0D50X00009XkgZFSAZ/problem-with-cubemx-generated-code-uart-intmode)

gkasprow, 2019-09-06:
So, the final SPI configuration is as follows:
    AUX: SPI1: 32 bit, clocked with SPI123
    ADC0: master SPI2: 32 bit, clocked with SPI123, NSS shared with TIM4_CH4 on PB9
    ADC1: master SPI3: 32 bit, clocked with SPI123, NSS shared with TIM2_CH1 on PA15
    DAC0: master SPI4: clocked with SPI45, swapped MOSI with MISO
    DAC1: master SPI5: clocked with SPI45, swapped MOSI with MISO
    EEM: slave SPI6, independent D3, dedicated DMA and RAM

closed 2019-09-06 by gkasprow

jordens, 2019-09-06:
Yes. The io swap works.