sinara-hw/Stabilizer #17: DAC glitch/digital leakage

jordens, 2019-03-30:
I think I was seeing quite some update transients/glitching/digital feedthrough on the DAC outputs outside the 500 kHz bandwidth. A couple mV IIRC. I'll post shots if I find the time and if I can reproduce it. Did the AFE filter simulation include the non-ideal behavior of the opamps at higher frequencies?

gkasprow, 2019-03-30:
I used standard TINA models. I don't know how accurately it behaves at higher frequencies.
The simulations are [here ](https://www.dropbox.com/sh/3x5q54i9ryabo9d/AABXV05j78ZGPZhp7ULNxNXpa?dl=0)

jordens, 2019-04-01:
It was in fact not at the sample rate (500 kHz) but at one of the DCDC converter frequencies (400 kHz). 16 mVpp.

hartytp, 2019-04-01:
Good catch, but ouch!

jordens, 2019-04-01:
That was from the PoE module. I am using unshielded TP cables.
When powering it from the barrel connector there is still some other switching transient at 5mVpp and 500 kHz which I can't locate.

gkasprow, 2019-04-01:
I'm not sure if I still have any Stabilized on the lab to check it.

hartytp, 2019-04-01:
@jordens do you still see any of this noise if you look at the DAC outputs on the SMA?

jordens, 2019-04-01:
Sure. That's where I generally look.

gkasprow, 2019-06-25:
I took the Stabilizer home. I supplied it from the power bank to make sure there are no ground loops.
The output is the following:
[image]

gkasprow, 2019-06-25:
In my case the interference period is roughly 25us.

hartytp, 2019-06-25:
Which smps is that? Is that at dc output of 0V?

gkasprow, 2019-06-25:
That's funny. I switched the power off. And the signal looks similar.
[image]

dnadlinger, 2019-06-25:
> That's funny. I switched the power off. 

Switched the power off, as in the power to the circuit and it's actually external interference still?

gkasprow, 2019-06-25:
there are ~2.8us spikes visible after I trigger the scope with <5ns pulses and turn averaging.
But none of the converters on the board works with such frequency
[image]

gkasprow, 2019-06-25:
when I switch the power off, the bit interferences stay, but these small spikes go away.

gkasprow, 2019-06-25:
All power rails are quiet, only P3V3 have 20mV ripples, but not correlated wit the output

gkasprow, 2019-06-25:
[image]

gkasprow, 2019-06-25:
It looks like these spikes are caused by me. When I touch the board they appear, when I don't, they disappear.

gkasprow, 2019-06-25:
they are caused by the power bank. When I connect only GND with power bank, they appear.

gkasprow, 2019-06-25:
but they are induced only on the scope probe. When I touch the board GND,  at the crocodile clip, they are also there. SMA cable doesn't help much because we have CMC which has limited attenuation rate for such short pulses.

hartytp, 2019-06-25:
Wrap a ferrite on your power and scope cable?

gkasprow, 2019-06-25:
True, this could help for such a noise source.

hartytp, 2019-06-25:
Worth a go.

Anyway, pick-up from external SMPSs is always an annoying issue.

hartytp, 2019-06-25:
The bigger issue I saw -- and the one that concerns me most -- was the noise on the DAC output that scales with the programmed DAC voltage. See here https://github.com/sinara-hw/Stabilizer/issues/9#issuecomment-499091392

As this noise scales with the DAC voltage, it's definitely not due to an external source. Were you able to reproduce this?

hartytp, 2019-06-25:
> they are caused by the power bank. When I connect only GND with power bank, they appear.

I've lost track. In your measurements did you see any noise at all on the DAC output that wasn't caused by pickup from external sources?

gkasprow, 2019-06-25:
It looks that all I see comes from external sources. The noise you mentioned seems to be covered by them. I thought I'm living in a quiet spot. It is quiet, but not from EM point of view. Tomorrow I will connect it to the 12V battery directly and try to reproduce it.

hartytp, 2019-06-25:
what DAC dc output voltages did you measure?

I guess it’s possible that what I’m seeing is just filtered amplified pick up that’s getting in to the dac ref circuit. Although if that’s really the case then it’s odd that I see so much more noise on the DAC than the ADC.

hartytp, 2019-06-25:
I’ll try to find a screened box for stabilizer and take some more data.

Can you remind me of a on for a screened Eurocrats box that I can mount stabilizer in with its panel?

gkasprow, 2019-06-25:
I observed default DAC output which is 0V.

hartytp, 2019-06-25:
@glasprow ok. No the noise I see scales with the DAC output. If the output is 0V then you don’t see any noise. So you have to use a fw that programs it to full scale or something...

gkasprow, 2019-06-25:
OK, I will have a look it tomorrow.

hartytp, 2019-06-25:
:)

gkasprow, 2019-06-25:
Do you have some simple binary that outputs max value? I don't want to mess up with setting toolchain.
If you could provide me binary I could load via DFU that would make my life simpler.

hartytp, 2019-06-25:
@gkasprow sure, I'll send you something tomorrow.

What I will do is change this one line of @jordens firmware to set the DAC to a fixed value of `0xffff` https://github.com/quartiq/stabilizer/blob/d9aa3f440143f0567c1259b85b3051eef24cd349/src/main.rs#L806

@gkasprow I haven't flashed using DFU, but OpenOCD worked fine (the only slightly annoying thing was finding a recent version).

Generally, I was amazed how easy it was to build and flash the firmware. Following Robert's instructions and the [embedded rust book](https://rust-embedded.github.io/book/) it took me about half an hour to install Rust from scratch, build and flash for the first time. The Rust tooling is very impressive and makes this kind of thing very easy. I'm happy to send you a binary, but you may find it faster to make your own...

gkasprow, 2019-06-26:
I usually use IDE tools for C language under Windows (Atollic IDE) and STM32CubeMX. Never used RUST.
I have a virtual Linux machine but it always takes some time to set it up.

gkasprow, 2019-06-26:
I'm looking at the DAC reference input and it's fine.

hartytp, 2019-06-26:
@gkasprow here is a hacked version of @jordens fw. On power on, it sets DAC 0 to +fs and DAC 1 to -fs. I checked on a scope and I can clearly see the oscillations I posted before.

[release.zip](https://github.com/sinara-hw/Stabilizer/files/3329391/release.zip)

gkasprow, 2019-06-26:
@hartytp I took your files. It is elf output. I converted using 

> arm-none-eabi-objcopy -O ihex

then converted to dfu and CPU doesn't start.

gkasprow, 2019-06-26:
Actually it does something. both DACs are now set to -10V. And still no issues.
[image]

gkasprow, 2019-06-26:
This time I supplied it from linear bench supply.

hartytp, 2019-06-26:
@gkasprow one DAC should be at +10v and one at -10V. Maybe it's not flashed correctly? Try following the instructions at https://github.com/quartiq/stabilizer for how to flash via DFU

It's possible that the noise is related to CPU activity so if the CPU isn't running then maybe the noise is gone (although, halting execution from GDB didn't make an impact on the noise I saw so I don't think that's the case).

gkasprow, 2019-06-26:
It's quite possible that it comes from the CPU.
I followed instruction, I do the verification.
[image]

[image]

gkasprow, 2019-06-26:
Ethernet makes huge difference
[image]

hartytp, 2019-06-26:
All the tests I did were with Ethernet connected btw.

gkasprow, 2019-06-26:
did you set the option bytes?

hartytp, 2019-06-26:
> It's quite possible that it comes from the CPU.
> I followed instruction, I do the verification.

Interesting. I'll re-flash my board with DFU tomorrow and check that the files I posted do what I think they should :) I'll also see what happens when I remove the ethernet connector

hartytp, 2019-06-26:
> did you set the option bytes?

?

gkasprow, 2019-06-26:
the CPU has a FLASH region that can be programmed called Option Bytes.

gkasprow, 2019-06-26:
It looks like these interferences go through the output CMC.
When I short pins 9, 10 of the AFE connector, the output gets clean.

hartytp, 2019-06-26:
> the CPU has a FLASH region that can be programmed called Option Bytes.

Pass. Not sure

hartytp, 2019-06-26:
> It looks like these interferences go through the output CMC.
When I short pins 9, 10 of the AFE connector, the output gets clean.

Good to know. I'll try that tomorrow as well.

What's your model here? That the CMC picks up noise? Or that having the DAC SMA ground floating at AC is a bad idea for noise?

jordens, 2019-06-27:
https://github.com/sinara-hw/Stabilizer/issues/9#issuecomment-499116829

gkasprow, 2019-06-27:
No, CMC is just a transformer. It's good at removing the common noise, but it does well under one assumption that both lines have the same load impedance. Which is not true in our case.
The CMC part that is connected to GND breaks the ground loop created by Ethernet switch, PoE transformer capacitance, oscilloscope, etc. this loop has low impedance for high frequencies, that's why we observe short spikes. The CMC part that is connected to the signal sees only 1M scope impedance.

jordens, 2019-06-27:
What does that mean for the cmcs on zotino, sampler and idc2bnc? They are also typically used across severe impedance mismatches.

gkasprow, 2019-06-27:
It strongly depends on the particular application. Maybe we should add an option to bypass the CMCs. Or even not mount them and provide small PCB with CMC and SMA connectors.
In Sampler the CMC is loaded in a symmetric way. In Stabilizer case, the voltage induced by the ground loop is even 1V, so the CMS still does a good job removing it. However, due to an uneven load, the process is not optimal.

jordens, 2019-06-27:
What would a proper "common mode ground loop suppressor" between a low impedance source (a DAC) and a high impedance sink (a trap electrode) look like? Would that need a CMC with a a turns ratio as the sqrt of the ratio of impedances?

jordens, 2019-06-27:
Leave the option bytes at their default values.

gkasprow, 2019-06-27:
The only valid turn ratio is 1:1, otherwise, it won't work as CMC.
The CMC has two impedances, common and differential. Common is usually in uH..mH range, the differential is in nH range. If the signal is passed in a symmetrical way, nothing happens because we have the same current in both windings and the signal sees only differential inductance. But when the circuit is asymmetrical, one winding will get higher current and another will get much smaller which means that CMC is no longer serving the purpose.  The signal will see the common inductance instead. If we terminate the DAC after the CMC, at least for high frequencies, the situation would vasty improve.

jordens, 2019-06-28:
Interesting. I.e. a parallel R-C with the same R as the output R of the DAC/driver? Would be nice to see that tested.

hartytp, 2019-06-30:
> It looks like these interferences go through the output CMC.
>When I short pins 9, 10 of the AFE connector, the output gets clean.

> #9 (comment)

Interesting. @gkasprow I believe the issue you're seeing here is entirely separate to the issue I saw. Firstly, the noise spectrum is different. Beyond that, I also tried both PoE and an external supply, as well as removing the ethernet. I do not recall any of this making a significant difference (but will double check next week). Given all that and the fact that the noise scaled linearly with the DAC output voltage, I don't think that what I saw was a grounding issue.

I'll try shorting the CMC next week to be sure.

It's interesting you weren't able to reproduce my findings, but maybe not surprising since the DAC output voltages were not correct during your tests. I'll re-flash the binaries I posted next week and take a few measurements to see if I can figure this out.

hartytp, 2019-06-30:
> What does that mean for the cmcs on zotino, sampler and idc2bnc? 
> ...
> It strongly depends on the particular application.

Indeed. That's the problem with these grounding issues -- it's hard to find a solution that works in all cases.

FWIW on all the tests I've done on the other boards, I've never seen the CMCs do any harm.

> Maybe we should add an option to bypass the CMCs. 

Can we fit jumpers across the AFE headers? Ship the board with the jumpers in and allow users to remove them as desired?

> What would a proper "common mode ground loop suppressor" between a low impedance source (a DAC) and a high impedance sink (a trap electrode) look like?
> ...
> If we terminate the DAC after the CMC, at least for high frequencies, the situation would vastly improve.

Yes, for a trap DAC, it's probably best to use RCs to make a symmetric differential AC termination.

[9 more replies]

closed 2019-09-06 by gkasprow