sinara-hw/Stabilizer #19: New revision - release date

jmilczarek, 2019-04-16:
When should we expect a new release of Stabilizer? We need to produce two more Stabilizer boards, but maybe a few more pieces would be better. I wonder if 1.0 version would be marketable in the next months.

gkasprow, 2019-04-18:
there are no serious issues and boards are currently under tests so it's too early for a new release.
make sure you do not solder BD5230 (IC10).

hartytp, 2019-06-06:
I think once we've figured out the source of the excessive noise on the DACs and made a decision about the ADC buffers, we should consider moving to a new revision.

hartytp, 2019-07-05:
I think it's time for a new revision.

hartytp, 2019-07-12:
@gkasprow what's the timeline for the next revision?

gkasprow, 2019-07-12:
I will be back from the US in mid-August, then have 2 weeks of holidays, so should finish it at the beginning of September, maybe earlier. It depends on weather in Poland :D

gkasprow, 2019-08-19:
I'm working on it right now. Expect it in a few days.

closed 2019-08-19 by gkasprow