sinara-hw/Stabilizer #20: [RFC] New daughterboard: Piezo Driver

dtcallcock, 2019-05-16:
I am thinking that many Stabilizers will end up servoeing on piezos (doubling cavities, laser cavities, mirror adjusters etc.).  Therefore does it make sense to have a dedicated daughterboard for this?  

Obviously [Zapper](https://github.com/sinara-hw/zapper/issues/1) is under development but may end up being overkill for this application.  I'm imagining a very simple board with a DC-DC module, a couple of HV opamps and a few DIP switches to set ranges.

gkasprow, 2019-05-16:
The student who wants to build HV drier just reactivated so that we can simplify his life and rearrange the Zapper concept a little bit.
Is such driver usually used in a crate or instead as a stand-alone box?

dtcallcock, 2019-05-16:
My feeling is that Zapper would be used in a crate as part of a larger system (and have other use cases  too like trap electrode voltages).  This daughterboard would be something that would allow Stabilizer to be standalone and still drive piezos.

hartytp, 2019-05-16:
What bandwidth + load capacitance does this board need to drive?

Also, might make sense to put a power connector on the FP that can power a small transimpedance amp that goes right by the photodiode (maybe a small connector with signal and power?).

dtcallcock, 2019-05-17:
My immediate use case would be to drive a pair of [Fabry Perot etalons](https://www.thorlabs.com/thorproduct.cfm?partnumber=SA200-3B).  This would replace two [$1k Thorlabs boxes](https://www.thorlabs.com/thorproduct.cfm?partnumber=SA201) and a data aquisition card.

The piezos in the Fabry Perot have a capacitance of ~500nF.  Those driver boxes put out 0-45V, 15mA, can do 30V/ms slew, and have 1mVrms (6.6mVpp) output noise.  So ideally this board would match or beat those specs.

The ability to put out up to 0-135V would probably maximize wider applicability.

It looks like an [LTC6091 dual opamp](https://www.analog.com/media/en/technical-documentation/data-sheets/6091fa.pdf) and [COTS power supply](https://www.mouser.com/ProductDetail/RECOM-Power/R12-100B?qs=sGAEpiMZZMsc0tfZmXiUnQxvSVjX95ZuDYch3%2F1dutc%3D) would do the job.

dtcallcock, 2019-05-17:
> Also, might make sense to put a power connector on the FP that can power a small transimpedance amp that goes right by the photodiode (maybe a small connector with signal and power?).

We generally use [M8-3 connectors](https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/T4043014031-000/A126826-ND/7927425) to match the ubiquitous [Thorlabs photodiode modules](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=2655).

Would be nice to have and there will be plenty of panel space.  They are a little pricey if people are unlikely to use them though.  The daughterboard should at least have a header for these though.

gkasprow, 2019-05-21:
There are compatible low-cost plastic M8-3 connectors available from multiple vendors.

adamkolodynski, 2019-05-24:
Hello, my name is Adam Kołodyński, as my B.SC thesis will be a part of this project, I would like to confirm the main goals which I have to achieve. In a previous thread, I saw the need for 0-150V output, and a reasonable bandwidth would be 75kHz. I think using Zotino would make my project more efficient.

dtcallcock, 2019-05-24:
Presumably you are talking about [Zapper](https://github.com/sinara-hw/zapper/issues/1).  It probably makes sense to finish specifying Zapper on that thread before deciding whether it still makes sense to design this proposed Stabilizer daughterboard.

sbourdeauducq, 2019-06-24:
> My immediate use case would be to drive a pair of [Fabry Perot etalons](https://www.thorlabs.com/thorproduct.cfm?partnumber=SA200-3B). This would replace two [$1k Thorlabs boxes](https://www.thorlabs.com/thorproduct.cfm?partnumber=SA201) and a data aquisition card.

We can see here that the name "Stabilizer" is poorly chosen if that board is going to generate signals that scan a Fabry-Perot interferometer (which may or may not be used to stabilize a laser later on).

This is a different set of skills, but it might be interesting to make our own SFPIs as well. Maybe with this kind of mirror coating (if that's usable at all?) so we can have a wide wavelength range: https://www.newport.com/f/ultra-broadband-dielectric-mirrors
(it does not seem to be available on spherical mirrors, unfortunately)

> It looks like an [LTC6091 dual opamp](https://www.analog.com/media/en/technical-documentation/data-sheets/6091fa.pdf) and [COTS power supply](https://www.mouser.com/ProductDetail/RECOM-Power/R12-100B?qs=sGAEpiMZZMsc0tfZmXiUnQxvSVjX95ZuDYch3%2F1dutc%3D) would do the job.

FWIW I've been playing with high-voltage high-speed broadband amps using discrete components and got some interesting results. Some schematics and data are here: https://github.com/m-labs/spectropak/tree/master/hvamp_proto3
If I were to do this again I would try to use MOSFETs and not rare BJTs.
I can recommend against the Apex integrated amplifiers which are expensive, do not perform particularly well, and contain BeO.
If that LTC6091 meets the requirements, it looks like a very nice option.

gkasprow, 2019-06-24:
@sbourdeauducq did you build this circuit? I'm bit worried about the temperature stability of the bias current. Would such circuit be really much cheaper than Apex opamp?.
I would rather go for [HV opamp](https://www.analog.com/media/en/technical-documentation/data-sheets/ADHV4702-1.pdf) with unity gain power stage at its output. That approach would be far simpler.

[This](http://ww1.microchip.com/downloads/en/DeviceDoc/HV254%20C070813.pdf) is also interesting approach when one needs a lot of channels

dnadlinger, 2019-06-24:
Analog also recently released a new op-amp that works up to 220 V, ADHV4702. It's £12 in hundreds, so comparable to the LTC6091, with better drift and 1/f noise specs. LFCSP only, though, so you better make sure your PCBs are clean…

gkasprow, 2019-06-24:
would such ADHV4702 output current be sufficient?  It is rated for 20mA

dtcallcock, 2019-06-24:
For the Fabry-Perots, the driver boxes I want to replace can only do 15mA so these should work.

I don't have numbers for our doubling cavity piezos to hand but will look them up.  The drivers used in the NIST servo box are [OPA453](http://www.ti.com/product/OPA453) which can only do 50mA, so I'd be surprised if 20mA won't do the job.

dtcallcock, 2019-06-24:
> This is a different set of skills, but it might be interesting to make our own SFPIs as well. Maybe with this kind of mirror coating (if that's usable at all?) so we can have a wide wavelength range: https://www.newport.com/f/ultra-broadband-dielectric-mirrors
(it does not seem to be available on spherical mirrors, unfortunately)

I would worry that the Finesse would be an unpredictable function of wavelength if you used HR mirrors (>99% could mean anything).  I also think a confocal design is a must for these.

Layertec sell overrun mirrors (inc. curved ones) for a reasonable price if you only want one or two: https://www.layertec.de/en/shop/

What is driving your desire to make your own SFP.  The cost of the Thorlabs ones or you just want something more broadband?  If the latter, you need to consider the photodetector and not just the coatings.

sbourdeauducq, 2019-06-25:
> @sbourdeauducq did you build this circuit?

Yes. I did get hundreds of mA at 100V up to many MHz :)
(200Vpp even, with a pair of them)

> I'm bit worried about the temperature stability of the bias current.

In version 1, it was poor and resulted in [fireworks](https://www.youtube.com/watch?v=JCPXckfT-6g). On later prototypes, I did not measure it carefully, but it seemed fine. The four transistors at the output (TO-126 packages) are "sandwiched" with the heatsink and thermal sauce (T1 T2 heatsink T3 T4, with T2 and T3 being the power transistors) so they have similar temperatures.

> Would such circuit be really much cheaper than Apex opamp?.

For the parts cost, yes, definitely - especially if redesigned with widely available MOSFETs instead of obsolete Sanyo transistors rebranded and overpriced by NTE. It also has higher bandwidth and current.

> What is driving your desire to make your own SFP. The cost of the Thorlabs ones or you just want something more broadband?

Cost + wavelength range + ordering anything from Thorlabs is somewhat annoying.

gkasprow, 2019-06-25:
Switching to MOSFETs won't be so straightforward. They have high input capacitances and need low impedance voltage drive. It's easy for audio but challenging for 10MHz. Could be easier with SiC, but p-channel ones are not common. The question is if opamp + follower wouldn't be enough?

gkasprow, 2019-06-25:
I played a bit with Tina and simulated quite interesting approach to MOSFET power amplifier.
The idea is to drive MOSFETs directly from high current opamp in such a way that both DC path and AC path have identical gain.
In this scenario, one can scale up the amplifier to any power and keep optimal DC and AC performance.
This particular amp nave no problem to drive +/- 60V into the 1nF load at 1MHz. It is limited only with MOSFETs saturation current.
[image]

To limit the zero-crossing distortions, the additional bias current is injected to the current mirrors
[image]

If somebody wants to play with it, the simulation file is here:
[Piezo_drv.zip](https://github.com/sinara-hw/Stabilizer/files/3323711/Piezo_drv.zip)

gkasprow, 2019-06-25:
It can also be scaled to any voltage. The only limitation are the MOSFETs.

gkasprow, 2019-06-25:
In real life, one can use discrete current mirrors. Some low-ohmic resistors would be needed to make such current mirrors really working. Short-circuit protection will be needed as well.

gkasprow, 2019-06-25:
MOSFETs threshold have negative Th voltage coefficient so the compensation circuit may be needed.

hartytp, 2019-09-06:
This should probably move to a new repo...