sinara-hw/Stabilizer #22: AFE headers

hartytp, 2019-06-20:
The AFE headers on the boards supplied by Technosystem do not match the 3D model and aren't the same as the ones on the BOM. Moreover, the stand-off height is a couple of mm different for one of the headers.

If we're going to use these for mounting AFE mezzanines then the stackup height must be the same for all three connectors and must match the 3D model.

@gkasprow please double check the connectors you've used and add a note somewhere that they cannot be substituted for generic shrouded headers.

gkasprow, 2019-08-19:
@hartytp can you take photo of these headers?

hartytp, 2019-08-19:
I’m away this week but will do that next week when back.

hartytp, 2019-08-28:
@gkasprow I had a go at this, but it's hard to get a photo that clearly shows the issue.

Two of the headers seem correct (measuring the dimensions, they seem to match the data sheet for the part in the BOM). However, one is significantly (>0.5mm) different and ruins the board-board stackup. Even by eye, one can see that it's different to the others.

At a guess, I'd say that someone substituted one of them for a generic part number that they already had in stock. However, to allow reliable board-board stackup these must not be substituted and should be marked as such in the BOM.

hartytp, 2019-08-28:
Sounds about right.

gkasprow, 2019-08-28:
On the photo, the J3 looks different than other headers.

gkasprow, 2019-09-06:
All AFE connectors are 3M N25xx series so it is not an issue related with the project.

hartytp, 2019-09-06:
Okay, we just need to tell ts not to make substitutions unless parts are explicitly labelled as generic

dtcallcock, 2020-01-13:
Still an issue in v1.1

[image]

[image]

hartytp, 2020-01-13:
For now I suggest that the mating headers on afes like pounder are shipped unsoldered and soldered in situ by users

pkozakiewicz, 2020-01-14:
We usually use generic connectors for all Sinara boards except 2x15 IDC. Unfortunately somehow I haven't seen this issue yet, so our team took what we had in our warehouse and assembled it. I see that for now Stabilizer is the only board designed for board-board stackup, so we have to pay attention to it next time. I understand that every functionality works properly.

hartytp, 2020-02-18:
@gkasprow  Is there anything we can do here to avoid this issue in future? e.g. can we add a note on the BOM that these must not be substituted?

gkasprow, 2020-02-20:
Good idea to add the note to the BOM

gkasprow, 2020-10-10:
I added note "Use original! " to the BOM

closed 2020-10-10 by gkasprow