sinara-hw/Stabilizer #23: PoE turns on and off with Netgear GS308P

sbourdeauducq, 2019-06-26:
When connected to a Netgear GS308P switch, the board is powered for a few seconds, then the Ethernet link establishes, then the switch cuts off power, and the cycle repeats.
After a few cycles, sometimes the board stays powered on.

sbourdeauducq, 2019-07-10:
Problem is not present on a second board.

gkasprow, 2019-07-10:
It may be a result of slightly different capacitance tolerance causing a small difference in startup current

gkasprow, 2019-08-29:
Should I buy such a switch and try to solve the issue or we can assume that the user must use 30W power delivery? I don't have any stabilizer board in my lab.

hartytp, 2019-09-06:
What should we do about this?

gkasprow, 2019-09-06:
I'm not able to recreate the issue right now. All PoE switches I have in my lab work fine.

hartytp, 2019-09-07:
Can you think of anything that might have caused this on one of @sbourdeauducq's boards? Or do you think it was a defective board?

@sbourdeauducq can you send Greg the problematic board and switch?

sbourdeauducq, 2019-09-07:
I already sent the board back, the switch is in use 24/7.

hartytp, 2019-09-07:
Sounds like an isolated incident maybe with one board. Shall we close for now and reopen if we see this again?

sbourdeauducq, 2019-09-07:
As Greg explained earlier, it could be that the board is near the maximum for low-power PoE injectors and component tolerances put it on one side or the other of the limit.

hartytp, 2019-09-07:
Which component tolerances? What do we want to do about this issue?

sbourdeauducq, 2019-09-07:
If the solution is to reduce inrush current, we should fix it.

gkasprow, 2019-09-07:
But, the issue appeared once the Ethernet LEDs went on. It means it is not inrushed current of the DC/DC converters but current consumption of the Ethernet chip or something else which triggers the overcurrent protection.
I wouldn't do any design changes until we make sure what causes this issue.
The Stabilizer has multi-stage power sequencing and the 3V3 wakes up after 5V rail
So it looks like the OVC is triggered by +/-12V LDO, because it turns on just after the 3V3 rail.
The opamps may consume some excessive current before they reach working voltage.
Can you connect the scope to +12V input rail, +3V3, +- 12VA and +5V ?
Trigger it on falling +12V and see what rail causes the issue.

hartytp, 2019-09-07:
@gkasprow do you have a stabilizer card? If so, this is easy isn't it? Replace L6 with a 0R1/0R01 resistor and look at the voltage across that resistor with a differential probe while you power stabilizer on. That will tell you the current v time profile during startup. If that's marginal according to the PoE specification then we should fix it, if not I think we can assume that there was an issue with the specific board that @sbourdeauducq had.

dtcallcock, 2019-09-07:
How important is it that POE is supported by Sinara boards given POE+ kit doesn't seem that hard to come by?  For example that GS308P (8 port | 4xPoE 53W) is a $70 switch but the GS108LP (8 port | 8xPoE+ 60W) switch is only $85.

hartytp, 2019-09-07:
It's a nice to have, as it saves on the external clutter of having PoE splitters. It's particularly useful for small modules like Thermostat where one might want to get them into a tight space with limited cabling access.

gkasprow, 2019-09-07:
The question is about scrapping POE(802.3af) and support only PoE+ (802.3at) switches

dtcallcock, 2019-09-07:
This ^

hartytp, 2019-09-07:
aah, ok.

While I don't feel too strongly, I'd prefer to keep 802.3af support unless it turns out to be particularly hard to do. The reason being that it's likely that users will get confused about the distinction and use the wrong PoE switch type anyway.

gkasprow, 2019-09-07:
The design supports, but @sbourdeauducq found a single case where it doesn't work. So, if we want to support standard PoE, one needs to debug it.

hartytp, 2019-09-07:
Replaced L6 with 1R0 to monitor current transients during power on. Powering from external 12V supply via barrel connector.

[image]

hartytp, 2019-09-07:
Peak is 690mA. So only 8W

[image]

hartytp, 2019-09-07:
Longer time scale 
[image]

hartytp, 2019-09-07:
even longer 
[image]

hartytp, 2019-09-07:
So, no sign of anything bad there. GS308P is specified for 15.4W per channel and our peak power is just over half that.

[image]

I'll try again with a PoE switch to be sure.

hartytp, 2019-09-07:
Now with a netgear prosafe jgs516pe

[image]

[image]

hartytp, 2019-09-07:
So, based on that, I think we can rule out inrush currents from being an issue on Stabilizer.

Unless anyone can think of anything else that could be the issue here and ,given that we cannot reproduce this issue and we no longer have access to the board it manifested on, I suggest we close this for now and repoen/debug if it occurs again. Any objections?

hartytp, 2019-09-07:
hmm...one possibility here is that the real issue is the reference buffer oscillating on your board (due to the lack of decoupling). That's really marginal and is completely unstable on some boards. That can sink a hell of a lot of current into the decoupling caps.

Anyway, we've done some work on rail stability in general, so maybe that will help...

hartytp, 2019-09-08:
Nb if you have cheap Ethernet cables with high resistamce you’ll have issues like this due to loss in the cables during the turn on current spikes

hartytp, 2019-09-10:
Okay, closing for now. Let's reopen if anyone is able to reproduce similar symptoms, or if anyone can think of any further tests we should do

sbourdeauducq, 2019-09-11:
I am using high-quality short Ethernet cables and I had already tried to replace them, without effect. They also power other devices correctly.

hartytp, 2019-09-11:
@sbourdeauducq noted. But, without access to hardware that displays this issue, I don't see any good way of debugging it further. e.g. from the sounds of it, this could have been a QC issue like a short on the PoE module rather than a design fault.

hartytp, 2019-09-11:
Note to self, the spec is 

[image]

=

hartytp, 2019-09-11:
hmm...it's possible that the issue here is that we're right on the edge of the PoE module's minimum load specification. cf #50. Again, however, hard to tell witout hardware (if we had it, we could add a load resistor to the board and see if that fixed the problem).

hartytp, 2019-09-11:
We're nowhere near maximum current. Assume Vi = 37V, efficiency=0.8, Iout_peak=700mA. Peak input current is then 280mA, which is comfortably within spec.

hartytp, 2019-09-11:
Anyway, from a design review of the PoE design, the only potential issues I can see are the min load and the missing ESD diode. Other pairs of eyes always welcome!

sbourdeauducq, 2019-09-18:
@gkasprow I can send the switch now. Technosystem have the problematic board already.

gkasprow, 2019-09-18:
OK, let's sort this out before next  revision

gkasprow, 2019-10-21:
I got the board. I also have a switch but it works. The switch is, however, AF/AT. I ordered Netgear GS308P, the same @sbourdeauducq has. It is AF only.

hartytp, 2019-10-21:
I can reproduce this on a thermostat board. Interestingly, the first time I plugged it into PoE I got a green light on my switch. After that, each time I plugged it in, I get an error light and the power cycles on/off. Powering from a barrel connector it seems fine (5V, 3V3 and 3V3A all have the correct voltage).

@gkasprow what measurements should I take on this thermostat board to help narrow down the issue?

hartytp, 2019-10-21:
@gkasprow on this Thermostat, the PoE switch error LED blinks on/off for a while before eventually giving me a green LED. Someone (you??) kindly soldered antennae to the PoE rectifier diodes to make them easy to probe. Screen shorts below

hartytp, 2019-10-21:
PoE + and PoE -  (all using 10x DC-coupled probes)

[image]

Same but zoomed in time scale

[image]

hartytp, 2019-10-21:
Same but channel 2 now looking at the +12V output of the PoE module:

[image]

[image]

[image]

In the last shot you can see the power come on and stay on

gkasprow, 2019-10-21:
I think we have to look at the current. For such measurements, I use battery-powered scope

hartytp, 2019-10-21:
hmm...I think some of the photos above look odd because the scope probe came disconnected and so was basically AC coupled. I've got better photos I'll post later.

> I think we have to look at the current. For such measurements, I use battery-powered scope

That was my thought too. But, it's not totally straightforward since there is no easy place to probe (will probably have to life a rectifier diode and solder a sense resistor between the diode and its pad) and I'll be measuring a small signal on a large offset. I'll try that, but won't be for a couple of days.

What I may have time for later this eve would be to try adding a load resistor to the 12V output. That could tell us if it's the switch being fussy over minimum loads.

gkasprow, 2019-10-21:
I will play with it once I get a switch with 15W power delivery. All the switches I own are 30W ones

hartytp, 2019-10-21:
Cutting the PoE+ trace and adding a 100Ohm sense resistor seems to "fix" the issue. No idea why yet. Vsense=1.5V, so input current is 15mA or approx 75mW.

hartytp, 2019-10-21:
So, it sounds like this may just be a surge current issue. I bet that if I was using a long/cheap ethernet cable I wouldn't be able to reproduce this. The EMI filter in the next revision will almost certainly do the job.

I'll try a 10Ohm sense resistor and see if I can still reproduce the issue. But, it may be that any resistor large enough to give me a sensible reading is also large enough to stop the PSU from tripping...

hartytp, 2019-10-21:
Still boots reliably with 10Ohm

hartytp, 2019-10-21:
Probing across my sense resistor, I can't see any transients. So, maybe there is some funny instability that only occurs when the input resistance to the PoE module is too low.

FWIW adding a 150R load resistor to the 12V output didn't help, suggesting that it really is an issue with the input, not a violation of the minimum current.

Putting a 0R resistor back in and the problem promptly returns. 

A longer ethernet cable didn't fix the problem.

hartytp, 2019-10-21:
NB the resistor I added was between the PoE module pin 1 and the diodes.

@gkasprow I wasn't able to find a sense resistor largel enough to give me a current signal I could measure easily, which didn't make the issue go away...

What do you think about all of this? Any more measurements we should take? Or, shall we just hope that the EMI filter resolves the issue, but leave pads for some small damping resistors between the module and the diodes just in case.

The only issue is if we want to draw close to max power: 25W at 50V is 0.5A, so even 1Ohm would be 0.5W! Let's just hope the EMI filter fixes the issue!

hartytp, 2019-10-21:
@gkasprow what do you think?

hartytp, 2019-10-21:
FWIW a 22uH coil craft chip inductor (DCR~11Ohm) also fixes the issue. So, I'm now relatively optimistic that this will not be present in the new revision with the EMI filter.

I am, however, concerned about the power handling of the EMI filter on Thermostat, but that's a separate issue.

@jordens @gkasprow any objections to closing this issue? 

@gkasprow we really need to get the next batch of Stabilizer sent to manufacture this week. AFAICT, so long as no one has any more concerns about this issue, there is very little to do. Would you mind finishing off and preparing for manufacture asap please? Thanks!!

hartytp, 2019-10-21:
> I will play with it once I get a switch with 15W power delivery. All the switches I own are 30W ones

FYI My switch is 30W as well. For some reason I can only reproduce this on some switch channels on some boards, so I think it's something being very marginal. Anyway, I think I now have all the data we need on this...

gkasprow, 2019-10-21:
OK, let's close and hope this will resolve the issue.

hartytp, 2019-10-21:
Ok. As I said, if you think of any more measurements we should take, let me know.

gkasprow, 2019-10-23:
I received the Netgear GS308P switch and I can confirm - the Stabilizer turns on for 3-4s, then turns off for another 3s and the cycle repeats. Now, I can debug this issue.

gkasprow, 2019-10-25:
I placed 5Ohm resistor in series with PoE converter and it starts just fine. There is 0.3V drop on it.

gkasprow, 2019-10-25:
I tried with the 100nF capacitor between G and S of T5 but it does not help. It looks like it's the PoE converter that causes a current spike.

hartytp, 2019-10-25:
> I placed 5Ohm resistor in series with PoE converter and it starts just fine. There is 0.3V drop on it.

That's roughly consistent with my observations. I found it hard to pick a small enough resistor to avoid interfering in the dynamics, while still getting a measurable signal.

What did the transients look like? 300mV on 5Ohm = 60mA. 

Can you try putting an inductor in series with the PoE module? AFAICT, this also fixes the problem. It would be nice to see the current transients when the fault occur (although, that's hard to measure). But, AFAICT, any input filtering will resolve the issue.

gkasprow, 2019-10-25:
1R resistor is too low.

[35 more replies]

closed 2019-10-26 by hartytp