sinara-hw/Stabilizer #24: DAC noise

hartytp, 2019-07-05:
Breaking the DAC noise I reported in https://github.com/sinara-hw/Stabilizer/issues/9 out into its own issue:

With DAC 0 at +10V I see the following (AC-coupled) output driving a 1M scope:
[image]

- I see approx 7.5mV of noise at 140kHz
- This is with Stabilizer powered from a 12V supply, but I see the same with PoE
- No difference with ethernet + JTAG connected or disconnected
- No difference when I place a jumper across pins 9 and 10 of the analog header to short DAC_0_N to ground
- The noise I see scales with the DAC output voltage, so this seems like it must be noise on the reference. I'll post shots of different output voltages in a moment
- @gkasprow had a go at reproducing this in #9 but I do not yet trust his measurements since he wasn't getting the correct DC output. @jordens have you seen this noise in any of your tests?

hartytp, 2019-07-05:
Moved the setup to a different room. The output looks much cleaner (far fewer SMPS spikes visible) but the 140kHz noise unchanged. So I don't think it's environmental/grounding related. The good news is that, other than this noise, the output seems pretty clean...
[image]

hartytp, 2019-07-05:
I've turned on some digital averaging on my scope to make the noise we're looking at a little clear...

channel 0 and channel 1 both at 0xffff (+10V):
[image]

0xBFFF (+5V):

[image]

hartytp, 2019-07-05:
0x7FFF (0V):

http://10.255.6.27/GetImage
[image]

hartytp, 2019-07-05:
-10V (0x0000):

[image]

hartytp, 2019-07-05:
So, it does seem that there is some noise on the reference that's getting scaled by the DAC.

NB this was all done with the quartiq/master firmware changing a couple of lines to manually set the DAC.

hartytp, 2019-07-05:
Indeed, the noise is on the reference. here, the yellow and green traces are the two DAC outputs, with the DAC code set to 0x0000. The blue is a 10x scope probe looking at teh 100nF reference cap by DAC pin 5, the grey is a zoom in of that. Lots of averaging to pull the signal out.

[image]

hartytp, 2019-07-05:
@gkasprow oops.... I think this is the chopper-stabilizer amp you're using to servo out the LT3042 offsets. Look at figure 3 on the data sheet. https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4528-1_4528-2.pdf

I'll depopulate some resistors to remove that reference buffer and confirm...

hartytp, 2019-07-05:
Indeed, that's the issue.

Same, but with R158 removed (inc heavy averaging)...

[image]

closed 2019-07-05 by hartytp

hartytp, 2019-07-05:
aah...no...removing R158 stopped the board from powering on correctly, as Rset for the LT3042 is only 39K. Replacing it with 41k allows the board to start and I see the same noise as before. So not that buffer after all...

hartytp, 2019-07-05:
I think it may be the LT3042 oscillating slightly. May not like such a capacative load.

The voltage on the set pin is clean, but the oscillations are clearly visible on the output.

hartytp, 2019-07-05:
Noise also visible on C139 (tantalum input capacitor). Maybe insufficient input capacitance. I'll try adding a large ceramic input capacitor in parallel with the 100nF.

hartytp, 2019-07-05:
Definitely an unhappy ref buffer. Definitely too much or too little capacitance somewhere...

gkasprow, 2019-07-05:
It didn't oscillate on my board.
Such LDOs tend to oscillate when too low ESR capacitor is used.
You can replace R94, R34, R105, R106 with 0.25R or even 1R and see if that helps
You can also check if IC27 is oscillating

gkasprow, 2019-07-05:
edit. Didn't read that you already removed the R158. So I'd try to increase these resistors or even remove them to see if that helps.

gkasprow, 2019-07-05:
5V is generated from SMPS, so check what is the frequency of it. But it should be far more than 140KHz. It is also passed via LDO so I'd be surprised if it was the source. Maybe IC16 is oscillating? Are the oscillations at the ref buffer input higher or lower than on the output?

hartytp, 2019-07-05:
Here is the P5V0 test point: 
[image] 

here is the LT3042 output:

[image]

hartytp, 2019-07-05:
So it seems that the LT3042 has almost no power supply rejection at the frequency. It's possible that the high-frequency PSRR is very low due to the heavy capacitive load.

hartytp, 2019-07-05:
> It didn't oscillate on my board.

A few mV is easy to miss when probing with a scope. So, it may have been there but hard to spot.

gkasprow, 2019-07-05:
I was observing it with 1mV/div scale and 20MHz BW.
I'd try with disconnecting the loads first.

hartytp, 2019-07-05:
hmmm... from the data sheet https://www.analog.com/media/en/technical-documentation/data-sheets/3042fb.pdf the PSRR should still be 80dB at 200kHz with 22uF output capacitance, so this doesn't seem at all right.

hartytp, 2019-07-05:
[image]

The PSRR drops off very rapidly with deceasing drop out voltage. However, on my board the 5V0A rail is at 4.975V, the ref output is at 4.3V. So, that's about 700mV head room, so I'd expect around 60dB.

I'll try disconnecting the loads and increasing C140 to 10uF (to ensure stability) next.

gkasprow, 2019-07-05:
I don't think it is caused by the input signal. The LDO oscillates dragging current from the input.

dnadlinger, 2019-07-05:
Ah, the minimum output capacitance is 4.7 μF – the regulator loop bandwidth is about 1 MHz, so it's possible the ESR/ESL to the ADC buffer caps is high enough to render the loop unstable with only 1 µF output capacitance.

hartytp, 2019-07-05:
@gkasprow  Yes. I believe that the output capacitance on the LT3042 is not sufficient and that's causing it to oscillate. It requires a 4.7uF ceramic output capacitor (ESR < 50mOhm and ESL < 2nH). However, we have a 1uF ceramic and then some 10uF tantalums quite far away.

Replacing the 1uF with 10uF the problem goes away.

@gkasprow the reason you didn't see this may have been that the circuit is boarderline stable so capacitor tolerances etc can cause it to move in/out of stability.

hartytp, 2019-07-05:
@dnadlinger indeed.

gkasprow, 2019-07-05:
This is this type of issue that appears once you produce more units...

hartytp, 2019-07-05:
Here is the outputs of channels 0 and 1, both set to 0xffff

[image]

I'll take more data in a quieter environment once i've reverted all my hardware modifications apart from the 10uF output capacitor.

hartytp, 2019-07-05:
Good news is that the output noise is now 1LSB RMS on both DACs, so it's actually a 16-bit system.

Same but different time bases

hartytp, 2019-07-05:
outputs look same at 0V as 10V. Good

gkasprow, 2019-07-05:
can you make FFT?

hartytp, 2019-07-05:
[image]

here you go.

gkasprow, 2019-07-05:
can you male a loop from crocodile GND clip and approach it to the SMPS converters and see if the 2 peaks on FFT are caused by them?
I'm also curious what this 8MHz peak comes from. Can you switch off the board and see if it is still there?

hartytp, 2019-07-05:
This is with heavy averaging on obviously.

There is a small SMPS peak at ~500kHz and a bump at 8MHz.

hartytp, 2019-07-05:
board off 
[image]

board on
[image]

hartytp, 2019-07-05:
closer in noise, board off 
[image]

board on 
[image]

So there is definitely a bit of 500kHz getting through somewhere. I'll take more data in a quieter room in a bit.

gkasprow, 2019-07-05:
can you disable CMC?

hartytp, 2019-07-05:
Shorted CMC:

[image]

unshort:
[image]

So in this particular case the CMC does make things ever so slightly worse

hartytp, 2019-07-05:
The 500kHz peak is still the same.

Anyway, there are still some noise peaks there, but pretty close to the noise floor. 

I'd have to check my noise model, but I guess we expect something like 15nv/rtHz output noise due to the 10k resistors, OpAmp etc. 

With stabilizer unplugged, I see a scope noise floor of 160uV RMS in a 20MHz BW (brick wall filter). With stabilizer I see 360uV. Implies stabilizer RMS in 20MHz is 330uV (scope noise is negligible). Assume noise roughly white. Noise floor is 70nV/rtHz. That's high. I'd expect more like 15nV/rtHz based on the OpAmp, resistors etc.
To put it another way

hartytp, 2019-07-05:
@gkasprow the 500kSPS is the DAC update rate, so that will be a little digital feed through sneaking in just above the noise floor.

gkasprow, 2019-07-05:
makes sense.

hartytp, 2019-07-05:
Putting it on a UPV audio analyzer with floating input, I measure 100nV/rtHz white noise (instrument BW is 250kHz). Instrument noise floor is moer like 10nV/rtHz.

Gain of AFE is 5, so that's about 20nV/rtHz in-band noise. That's about right given a 12nV/rtHz for the DAC, 6nV/rtHz for each OpAmp and 12nV/rtHz for each 10k resistor.

100nV/rtHz*sqrt(500kHz) = 70uV rms. So, still a bit lower than I measured. But, might be due to grounding etc. Anyway, I don't see anything particularly worrying here, so I'm happy to call a day on this.

hartytp, 2019-07-05:
Basically, this all works as well as we can hope...good!

gkasprow, 2019-07-05:
Great! Next well-engineered module thanks to the collaborative effort.

hartytp, 2019-07-05:
Note to self, to do at some point next week:
- put LT3042 OpAmp components back
- put LT3042 input capacitors back to original values (10uF tant + 100nF ceramic)
- check noise with PoE (CMC shorted and unshorted)
- replace/remove ADC buffers

gkasprow, 2019-08-19:
@hartytp what's the conclusion then? Which components should I change at the reference buffer?

gkasprow, 2019-08-19:
I found it. I should increase LT3042 output capacitance to at least 4.7uF