sinara-hw/Stabilizer #27: output cmc

hartytp, 2019-07-05:
we should make a decision about what to do here: leave it as is? scrap CMC? fit jumpers to short the CMC by default? add a differential output filter?

I'm happy with it as is, since I haven't yet observed any situation where the CMCs make things worse by a significant amount. I'm also happy scrapping since I haven't seen them make anything better yet.

gkasprow, 2019-07-05:
One can always add one outside of the board when it's necessary. I'm not sure if there are such ones with SMA connectors. We can always make a tiny PCB with CMC, RC pads and SMA connectors

jordens, 2019-07-12:
I would like to have them bypassed or even scrapped/DNPed by default. With PoE they make it significantly worse. sinara-hw/Thermostat#31 and #17 may or may not resolve this.

hartytp, 2019-07-12:
I don't have any objection to scrapping the output CMCs.

gkasprow, 2019-08-19:
added 0R mounted by default. CMCs are DNP by default.

closed 2019-08-19 by gkasprow