sinara-hw/Stabilizer #32: v1.1 changelog

gkasprow, 2019-08-19:
- moved SMA connectors 0.75mm to the front
- added PHY reset line (pin2 of CPU)

hartytp, 2019-09-05:
@gkasprow when do you think you'll be able to finish this release?

jordens, 2019-09-05:
* #16: Can we change the SPI bus assignment?
* #12: SPI buffer delay
* #11: Panel design. And move the USB connector not just 0.5mm but an entire mm. The plugs are long enough and I see enough space w.r.t. RJ45 jack.

hartytp, 2019-09-05:
There are a few other things as well. e.g. the reference buffer capacitor since we don't want that oscillating.

hartytp, 2019-09-05:
@gkasprow I'm keen to buy some more stabilizer cards asap, but would like to wait to the new release, so it would be great to get this ready asap

gkasprow, 2019-09-05:
@hartytp  I'm quite busy right now because 6 of my students decided to graduate without warning, so I have a lot of paperwork to do. I'll do my best to publish the release during the weekend.

hartytp, 2019-09-05:
They grow up so fast :)

gkasprow, 2019-09-05:
True :)

closed 2019-09-06 by gkasprow

hartytp, 2019-09-06:
Thanks @gkasprow ! I’ll review at some point this weekend and sign off

gkasprow, 2019-09-06:
I drafted the new release.

hartytp, 2019-09-07:
Okay, I've finished my review (everything apart from the digital bits).

Once @jordens signs off we can move to production. 

Great work all. Looking over the schematic again I was please to find how high quality it is. Really nice design :)