sinara-hw/Stabilizer #38: IC12 compensation

hartytp, 2019-09-07:
This looks a little different to the data sheet, since you've added the RC compensation to the FB pin. I don't have any reason to suspect an issue here but, as always, I'm curious :) Where did the design come from and are you definitely happy with the component values? If so, please close.

gkasprow, 2019-09-07:
The last row of the table
[image]

gkasprow, 2019-09-07:
I've just noticed that I used wrong inductance. It should be 4.7uH instead of 6.8. I simply copied this design from another project where it works already for years.

closed 2019-09-07 by gkasprow

hartytp, 2019-09-07:
Okay if you’re happy with that, it’s just odd to me that you’ve added extra compensation not mentioned in the data sheet. If you’re happy with t it’s fine by me though

gkasprow, 2019-09-07:
These compensation components are mentioned in the datasheet, in the table.
But you are right, it is not easy to find them.