sinara-hw/Stabilizer #42: IC1AA

hartytp, 2019-09-07:
Didn't we agree to scrap this buffer (output of ADC to microprocessor input) since it's not needed (microprocessor can work with 2V5 logic levels).

Maybe not such an issue now we have faster buffers, but this slows down the SPI bus, increases costs/power consumption.

gkasprow, 2019-09-07:
OK, Let's make them DNP and place 0R instead. If we discover issues then we will install the buffers.

closed 2019-09-07 by gkasprow

hartytp, 2019-09-07:
> If we discover issues then we will install the buffers.

Are you worried about noise from the microprocessor coupling onto the ADC? If so I'll open an issue to check that once the new hw arrives (will take 5min to check the RMS noise with a terminated input)