sinara-hw/Stabilizer #45: copy fastino filter design

hartytp, 2019-09-07:
Since this is ready in time after all, any objections to porting Stabilizer to it? Lower cost, lower power consumption, lower noise. More component reuse between designs.

dnadlinger, 2019-09-07:
The (simulated) transient response is currently considerably faster-settling on Stabilizer – there is probably some room for optimisation by fudging component values, though:

[image]

hartytp, 2019-09-07:
[image]

hartytp, 2019-09-07:
Are you happy with this as your optimized filter design?

hartytp, 2019-09-07:
The DAC settling time is 1us, so this isn't much slower than that. I'm happy with that settling time for Stabilizer personally.

hartytp, 2019-09-07:
(at least, I think the benefits of the new filter design outweigh the downside of a slightly slower response)

gkasprow, 2019-09-08:
@hartytp do we want to change the reference for Stabilizer DAC as well? We would need to add dedicated 5V one or switch ADC to 5V. It's a bit of extra work.

gkasprow, 2019-09-08:
We can simply replace DAC resistances with external SOT-23 matched resistor to limit the noise and leave the rest as it is now.

hartytp, 2019-09-08:
Yes I’d forgotten about the different reference. In that case I’d say the current afe is fine for stabilizer even if it’s not optimal. Let’s leave as is

closed 2019-09-08 by hartytp