sinara-hw/Stabilizer #46: SI

gkasprow, 2019-09-08:
[cf ](https://github.com/sinara-hw/Pounder/issues/9)

hartytp, 2019-09-10:
Do you plan to do this for the next release?

gkasprow, 2019-09-10:
Yes, I have a config file so doing it for the next revision is quick

gkasprow, 2019-09-15:
I meant SI/PI analysis :) Of course I will do both

gkasprow, 2019-09-15:
I added 50R source termination resistors on SPI lines, the lines are also 50Ohm ones. I don't know the driver resistance of STM32 chip, we can always change the resistors in case of issues

gkasprow, 2019-09-15:
6V rail
[image]

-13V rail
[image]

+13V rail
[image]

P12V0 rail
[image]

P3V3
[image]

gkasprow, 2019-09-15:
DC drop was well below 30mV including inductors for all rails.

closed 2019-09-15 by gkasprow