sinara-hw/Stabilizer #47: dead phy

hartytp, 2019-09-10:
@gkasprow we have the following observations:
- we had a stabilizer card that had been working fine for a while when powered from the barrel
 connector
- removed barrel connector and connected to PoE switch (netgear prosafe jgs516pe) to take some noise measurements. I've used this PoE switch before with different stabilizer cards without issue. Didn't check ethernet while it was connected to this switch
- connected back to an barrel connector
- ethernet phy seems dead (no link established)

@cjbe may have more details to add

Any idea what could have caused this? EMI associated with PoE? Should we post this card to you for investigation?

gkasprow, 2019-09-10:
Yes, post it please. I want to find what is going on before next release

hartytp, 2019-09-10:
Okay, good. I'll get it in the post to you in the am.

gkasprow, 2019-09-11:
you can short it temporarily with pin 14 which has pullup and see if that helps.

hartytp, 2019-09-11:
> @hartytp It could be an issue with dangling ethernet PHY reset pin.

It's internally pulled up isn't it? 

[image]

gkasprow, 2019-09-11:
OK, so this should work. I also added an external pullup resistor and was looking at the wrong schematic version.

hartytp, 2019-09-11:
Well, I'll post you the board back and you can see what you think

gkasprow, 2019-09-15:
Let's wait with production until I find what's wrong with Ethernet PHY.

hartytp, 2019-09-16:
@gkasprow did you get a chance to look at the phy?

gkasprow, 2019-09-16:
Not yet. I will be in my lab tomorrow.

gkasprow, 2019-09-19:
I replaced the PHY chip and it works.
No idea how you managed to kill it.
The chip works partially, at least its digital interface.
It generates the distorted heartbeat pulses on both Rx and Tx, but they are much longer and only one long negative, then one long positive. But it should be one long negative, then one short positive-negative and one long positive. It looks like the driver was not able to drive short pulses.
The TVS is OK.
Didn't you short the Ethernet connector leads with some potential?

hartytp, 2019-09-19:
@gkasprow okay, that's what I guessed.

We plugged the ethernet into our PoE switch, but that's about it. Possible someone touched the phy, and gave it an ESD shock (although I do not think so and, anyway, it has protection diodes).

Are you sure we have all the TVS needed to protect from external ESD, phy abs max voltage ratings are respected, etc etc

hartytp, 2019-09-19:
What should we do about this issue? If we can't see a reason for this phy being dead then it's hard to be confident that it won't happen again on a new board...

gkasprow, 2019-09-19:
The TVS are not very common in Ethernet applications and they are usually not required because transformer protect the PHY against common mode ESD voltage.

hartytp, 2019-09-19:
Do you have any theories for how this part could have been damaged (other than someone accidentally shorting something etc)?

hartytp, 2019-09-19:
If not, I guess we should close for now. FWIW from a quick skim over the schematic, I didn't see anything obviously wrong or any missing protection features.

hartytp, 2019-09-19:
Also, when you're done testing, can you post the stabilizer back (TS will send us some other things soon, so maybe they can ship everything together, it's not a rush).

gkasprow, 2019-09-19:
This is PHY connection diagram that was already used in numerous designs. It could be a faulty IC. It happens from time to time. The IC was broken but in the same way on both Rx and Tx channels. If it was ESD, one driver would be damaged.
The only reason for such PHY malfunction I see right now is 3V3 transient during on phase...

hartytp, 2019-09-20:
Well, I don't see any obvious issues with the 3V3 supply in this design, so shall we close the issue? Or, do you want to e.g. do a quite step response measurement to check the regulator is stable and happy?

hartytp, 2019-09-20:
> It could be a faulty IC

NB we used this board without issue for quite a while before connecting it to PoE, so the phy definitely wasn't dead when it arrived!

gkasprow, 2019-09-20:
Yes, but IC can break due to thermal stress after some time...

hartytp, 2019-09-20:
Would that be an issue with soldering then? Or a thermal management issue?

FWIW the 3v3 rail is more like 3v45 on one board I looked at due to the feedback resistors. But, that shouldn't be an issue, right? Abs max is 3v6

gkasprow, 2019-09-20:
I checked ADDV PHY and it looks just fine
[image]
On my board, the PHY supply is 3.40V

hartytp, 2019-09-20:
okay, fine. Let's close this and reopen if we see any issues on other boards. For now I'll assume this was most likely some form of user error

closed 2019-09-20 by hartytp

gkasprow, 2019-09-20:
Well, it's hard to say what could be an issue. A single failure does not give enough statistics :)

hartytp, 2019-09-20:
Thanks for investigating!