sinara-hw/Stabilizer #48: current probe resistor

hartytp, 2019-09-11:
@gkasprow In light of https://github.com/sinara-hw/Stabilizer/issues/23 what do you think of adding a pad for a 1206 sense resistor before the Vin+ input of the PoE module. Maybe also add test points on either side of the sense resistor to make probing easy. We would populate the resistor with 0R by default, but can mount a 2R0 or something if we need to debug any issues.

gkasprow, 2019-09-11:
Don't you want the CPU to know the power consumption? One low-cost opamp won't make a difference.

hartytp, 2019-09-11:
@gkasprow the main thing I want is an easy node I can probe to get a fast, accurate measurement of the current transients during turn on / turn off. An OpAmp is probably not ideally suited to that, although the sense amp from Booster might work (with the right pinout ;) ).

I'm less fussed about the CPU knowing the power, but I guess it could be useful. NB I'd like to be able to measure on the input of the PoE regulator not its output, so power calculation requires a measurement of both the voltage and current.

gkasprow, 2019-09-11:
OK, lest's stick with one resistor for now.
The CPU can also measure the current at the ground terminal without using Maxim chip.

gkasprow, 2019-09-15:
Added 0805 pads shorted by 0402 0R resistors to not increase the BOM.

closed 2019-09-15 by gkasprow