sinara-hw/Stabilizer #50: PoE minimum load

hartytp, 2019-09-11:
@gkasprow Am I missing something, or are we operating the PoE module at too low an output current (<200mA, see below) where it is not necessarily going to work due to the converter running in discontinuous mode.

[image]

[image]

https://github.com/sinara-hw/Stabilizer/issues/23#issuecomment-529145195

AFAICT this could be a particularly serious issue for Thermostat, where the load current could go to close to 0C if the TEC current was low.

hartytp, 2019-09-11:
Can it really be that this module can't function correctly without at least 600mW of load? If so, I'm not sure it's suitable for us (at the minimum, we'd need a switchable load resistor on thermostat to keep power enabled when the TEC current is low) :(

gkasprow, 2019-09-11:
yes, this could be also an issue with Sampler which does not consume much power when idle. I'm a bit surprised with that fact.

hartytp, 2019-09-11:
> Sampler 

You mean Stabilizer, right?

Anyway, yes, the measurements I posted suggest that the current consumption on the board I was looking (see link above) at is very close to 200mA. I didn't measure closely enough to tell if it's a little above or a little below (without taking a ruler to the screen shot!). But, it's definitely marginal, so this could be the problem. Probably also depends a little on what firmware has been flashed.

What do we do? Add load resistors or look for a different PoE module which is better behaved at low loads?

hartytp, 2019-09-11:
Ironically, using crappy cables may help here as it reduces the voltage at the input to the PoE module, increasing the converter duty cycle ;)

gkasprow, 2019-09-11:
[Other PoE+](http://www.farnell.com/datasheets/1820766.pdf?_ga=2.88353883.1301672071.1568192954-1090903098.1535327571&_gac=1.115033717.1566220327.CjwKCAjwkenqBRBgEiwA-bZVtlm9Qg_yAs5dOw4C9wMRxxjDIgu85B-jKR2io5e1SR0ckeA6OLFODRoCxwIQAvD_BwE) module do not have such constraint. The datasheet says something misleading about the max input inrush current that is also minimal current to maintain power signature requirment...

gkasprow, 2019-09-11:
But this module is too big to fit into enclosure.

gkasprow, 2019-09-11:
To solve this isue we can add something that consumes 200mA, for example fan :) It can be switched on by the CPU when necessary

jordens, 2019-09-11:
I remember point out the 200 mA minimum current before. But I cant find it. Back then IIRC the conclusion was that it should not be a problem. I haven't heard anything audible from those modules.

200 mA on 12V is 2.4 W, not 600 mW.

The MPS is 10mA at 20% duty, i.e. 2mA or 100mW average. We are always above that.

hartytp, 2019-09-11:
> 200 mA on 12V is 2.4 W, not 600 mW.

Indeed! 

> I remember pointing out the 200 mA minimum current before. But I cant find it. Back then IIRC the conclusion was that it should not be a problem. I haven't heard anything audible from those modules.

Okay, I don't recall that discussion. 

> The 803.2at MPS is 10mA at 20% duty, i.e. 2mA or 100mW average. We are always above that.

What's a good reference for these kinds of details on the af/at standards? 

Anyway, agreed, if that's the standard then we should be well above that. But, how do you interpret the statement in the data sheet then? (a) it's just wrong (b) it's an acknowledgement that many PoE switches are more fussy than the standard permits (c) something I've missed? @jordens if you're really happy this isn't an issue, feel free to close.

Anyway, from a review of the schematic/datasheets and based on my understanding of the relevant standards (which is limited), this was the only potential issue I could see.

jordens, 2019-09-11:
I don't have any special references.

My reading of the datasheet is that it we need to be above ~400mW (for 100% duty MPS including inefficiencies) and if we are below 2.4W then the module may make some noise. But I haven't observed that. I would not want to add an artificial load.

hartytp, 2019-09-11:
Okay. Well, let's close this for now then. If we find another board and switch combination that show issues then it will be easy to test if a load resistor fixes the problem and re-evaluate.

closed 2019-09-11 by hartytp