sinara-hw/Stabilizer #53: QSPI

gkasprow, 2019-09-29:
Consider routing QSPI to the extension connector instead of GPIO.
Another option is to add yet another high speed connector for 200Mbit QSPI link

hartytp, 2019-09-30:
> Another option is to add yet another high speed connector for 200Mbit QSPI link

If you can find something wit the right stackup, what about replacing all three connectors we currently have with a single Samtec connector (something like a lower pin-count FMC)?

hartytp, 2019-09-30:
Or, hell, go the full way and just use FMCs. i.e. make two front-pannels for Stabilizer: one which is the standard 4HP FP, and one with is 8HP with an AFE mezzanine cutout. Then put an FMC (or similar connector with fewer pins) and route all analog/digital lines on to that. 

That might actually work out nicer than the current design, since getting the stackup both electrically and mechanically reliable using separate FPs and pin headers can actually be a bit of a pain...

Thoughts?

gkasprow, 2019-09-30:
going this way, we can reuse Sayma AFE form factor and connector signal assignment...

gkasprow, 2019-09-30:
The issue with BGA connectors is that you cannot make such AFE at home...

hartytp, 2019-10-21:
@jordens what was your conclusion about all of this? What (if any) changes do you want to make for pounder?

jordens, 2019-10-21:
I think there is a pounder issue for this.

dtcallcock, 2019-10-21:
https://github.com/sinara-hw/Pounder/issues/10

hartytp, 2019-10-22:
Thanks for the link. Can someone more familiar with the microprocessor pin functions confirm what if any changes that requires to stabilizer please?

This is the last issue, so I’m keen to close ASAP and get v1.1 off to manufacture

jordens, 2019-10-23:
I guess @gkasprow will propose those changes. Connect QUADSPI to the AUX header instead of the SPI1 bus that is connected now. Drop LPUART1_TX and PC8 and use those for the two additional QUADSPI data pins.
After https://github.com/sinara-hw/Stabilizer/issues/13#issuecomment-528980208 it would then make sense to wire SPI1 to the EEM header instead of SPI6.

gkasprow, 2019-10-25:
@jordens we need both SPI and QSPI.
I will connect it instead of 
PD10
LPTIM1_OUT
HRTIM_CHE1
HRTIM_CHE2
HRTIM_CHA1
HRTIM_CHA2

I can also choose other pins instead

jordens, 2019-10-25:
Do we need both SPI and QSPI? I'd rather have timers available on the mezzanine.

gkasprow, 2019-10-25:
We need SPI for attenuators in Pounder and QSPI for DDS.
but, we can also connect it instead of 
LPUART1_TX
PC0
PD10
PC8
UART4 Rx
and one selected timer lines. 

The QSPI lines have also some useful alternative functions:
IO3
[image]

IO2
[image]

IO1
[image]

IO0
[image]

CLK
[image]

NSS
[image]

jordens, 2019-10-25:
OK. UART is less useful for the mezzanine IMO. We also have another header for a uart.
From my perspective either set of pins is fine.

gkasprow, 2019-10-26:
I mean that we already have this UART line among QSPI lines :)

gkasprow, 2019-10-26:
The header has the following pinout
[image]

gkasprow, 2019-10-26:
update

closed 2019-10-26 by gkasprow

hartytp, 2019-10-26:
Cool. Can you post a new rc pdf please?

gkasprow, 2019-10-26:
Sure, let me finish the routing. I also added QSPI source termination.

hartytp, 2019-10-26:
Ok. But if people request final changes to the schematic don’t you want to hear them before you do the routing?

jordens, 2019-10-26:
that pinout is fine from my perspective.