sinara-hw/Stabilizer #54: AFE connectors

hartytp, 2019-09-30:
What do people think about the following:
- Design two separate panels: a 4HP panel for use without an AFE and an 8 HP panel for use with the AFE mezzanine
- 8 HP panel has a standardized cutout for the AFE front panel (something like FMCs)
- Instead of having three separate 100mil headers, we have a single high-density connector for analog, digital and power

The motivations for this are:
- it can be tricky getting the stackup to work well with 100 mil pin headers and separate front panels. Using a Samtec connector and a panel cutout is likely to be more reliable mechanically
- the situation with three through-hole connectors is a bit of a pita for routing, since it really cuts up the power/ground planes and makes routing the boards hard. A single surface mount connector would take up much less space and be more robust
- having a higher density connector lets us route more IO to the mezzanines
- the current connector layout was designed to be vaguely compatible with the IDC->BNC boards, but that didn't really work out in a particularly meaningful way, so probably not worth the effort

Thoughts?

hartytp, 2019-09-30:
> The issue with BGA connectors is that you cannot make such AFE at home...

True. Well, we don't need anything like FMC density. So, let's choose something which is easier to hand solder, relatively cheap and available and somewhat robust, while being slightly higher-density than pin header.

Also, let's add lots of test points so that it's still easy to probe, despite being higher density!

(e.g. would be nice to have TPs for the microprocessor DACs and ADCs)

hartytp, 2019-09-30:
> going this way, we can reuse Sayma AFE form factor and connector signal assignment...

That's probably a bit overkill. e.g. we don't need 8 DAC and ADC channels. Let's go for something simpler

gkasprow, 2019-09-30:
A few cons:
- Dedicated cutout limits the amount of front panel connector area. It would have to be really big, but such makes the panel flexible. sub-panel fixing also takes precious area.
- 100mils connectors are very hard to break and easy to replace. Alignment is not a big issue, one must solder them mated.
- The high-speed connectors, especially SMT can be easily broken. 400pin FMC is quite robust due to the number of pins. 1mm or 0.8mm hand-solderable connectors go off easily when a lateral force is applied. Especially when there is some build-in tension caused by mechanical fixtures.  We could use rugged (and expensive) Harwin board to board connectors, providing that we find the right mating height.
- It's easier to place analog connectors on a mezzanine close to the Stabilizer SMAs, analog connector close to the CPU ADC and GPIO close to the CPU IO lines. Routing all these signals to common connector would need switchable termination or 0R options for high-speed ADC/DAC signals.

hartytp, 2019-09-30:
true

dtcallcock, 2019-09-30:
> the situation with three through-hole connectors is a bit of a pita for routing, since it really cuts up the power/ground planes and makes routing the boards hard

Could an SMT versions of 100mil header ([like this](https://www.mouser.com/datasheet/2/527/ssm_sm-1317000.pdf)) be used on more complex mezzanines where this is a problem without changing stabilizer design?

dtcallcock, 2019-10-09:
> It's easier to place analog connectors on a mezzanine close to the Stabilizer SMAs, analog connector close to the CPU ADC and GPIO close to the CPU IO lines. Routing all these signals to common connector would need switchable termination or 0R options for high-speed ADC/DAC signals.

What about a single SMT 100mil header that runs horizontally along the board so that one end is near the SMAs and one end near the CPU?  Right now we have 76 pins but could we get away with 72 and use something like [this](http://suddendocs.samtec.com/catalog_english/htss.pdf)?

If that's too long or we want to go to 80-100 pins we could drop down to 2mm or 0.05" pitch header (though I haven't checked the correct stackup height is achievable).  These are still generic, robust, and would allow for a mezzanine with a hand-solderable thru-hole header.

gkasprow, 2019-10-10:
well, this would involve the redesign of entire board. And this is a lot of work. And we would gain essentially nothing. With proper assembly technique, there is no issue with misalignment of the connectors.

hartytp, 2019-10-10:
Ok! I'm happy with the current solution! I just wondered if we should expose some more GPIO in case of future FPGA-based board, but it's probably fine as is. 

Shall we close this issue?

gkasprow, 2019-10-10:
we can always add another high-speed connector. I'd add it right now for QSPI signals.

hartytp, 2019-10-10:
> we can always add another high-speed connector. I'd add it right now for QSPI signals.

But we would need to find something that can be used at the same time as the current headers...

gkasprow, 2019-10-10:
that's not easy to find 18.7..18.8 stacking height connectors. 
But ERM/ERF series mates well even with the slight distance. 0.7mm won't make any difference to them.
Such connector could be placed somewhere on the board edge
[image]
The connector on the drawing is 5mm, but we would use 9mm one.
@hartytp what max frequency would you need in FPGA version? How many signals?
Maybe an additional 0.1" header would be sufficient?

hartytp, 2019-10-10:
> @hartytp what max frequency would you need in FPGA version? How many signals?

No idea, since I have no concrete plans for an AFE that would actually require this. More about leaving something for future expansion. If it can do the QSPI bus for pounder, it's a good start.

Out of curiosity, do you think that 250MHz is really too fast for pin-header? Getting 125MHz DDR down ribbon cable is easy. Is the issue here that it's LVCMOS rather than LVDS? If so, does the fancy connector make that much difference?

Basically, add it if it's helpful, but if it's not necessary don't bother :)

gkasprow, 2019-10-11:
250MHz is fine. 500MHz is probably fine, too. I used to pass USB 480Mbit/s signals via ordinary IDC and was working without issues.
So, let's add small IDC  connector for QSPI and other fast signals and I will close the issue.

hartytp, 2019-10-11:
Can't we just use the current "GPIO" header? Or, do we need more pins?

gkasprow, 2019-10-11:
That was my initial idea to replace the IOs with QSPI pins there. We don't need the QSPI FPASH memory on the mainboard anyway.

hartytp, 2019-10-11:
Sounds good to me unless anyone objects

Otherwise another header or high speed connector is fine by me as well

hartytp, 2019-10-21:
Sounds like there is no need to add an extra connector, so closing. feel free to re-open if anyone thinks an additional connector is necessary

gkasprow, 2019-10-23:
I will route the QSPI to existing connector and then close.

hartytp, 2019-10-25:
QSPI is covered in a separate issue, so closing

closed 2019-10-25 by hartytp