sinara-hw/Stabilizer #56: ERC fail

hartytp, 2019-10-04:
@gkasprow am I going crazy, or did you not run ERC? We were having some odd issues with the internal DACs, so I did some probing. The DAC reference voltage was at 3.7V (the 3V3 rail runs at 3V4 because of resistor choices, so this is clipping of an ESD diode). The 2V048 reference at C75 is fine, but the output of IC27B is railing. When I touch the input pin of that OpAmp with a probe the output voltage disappears for a moment...

Looking at the schematic, I don't see a black dot here...

[image]

The input is pin 5 of the OpAmp

[image]

but there is no connection visible on the layout

[image]

Did you forget to run ERC on this design ;)

@pathfinder49 I think this may explain some of the odd "firmware" issues you've been debugging

hartytp, 2019-10-04:
Not sure if this will have killed the CPU DACs.

hartytp, 2019-10-04:
@pathfinder49 we should be able to fix this easily by soldering a wire from that OpAmp to C75. Let me know if you want a hand doing that? Once it's done, check the output isn't oscillating and increase the 10R resistor if you need to to stabilize things.

gkasprow, 2019-10-05:
You are right. AD is even placing a red warning under the IC27B input.
I always do ERC, but for some reason, I skipped this one. No faintest idea of how this happened.

hartytp, 2019-10-05:
Np it’s an easy fix

gkasprow, 2019-10-25:
done. I also fived it on the board I got from @sbourdeauducq for debugging PoE

closed 2019-10-25 by gkasprow