sinara-hw/Stabilizer #62: Rc2 review

hartytp, 2019-10-26:
Thanks @gkasprow !

I’d like to send this to production on Monday afternoon. Please give feedback before then (or let me know if you need longer before we close this round).

hartytp, 2019-10-26:
- [ ] Why is Ic1B1C DNP?
- [ ] why is R124 undefined, not DNP?
- [ ] Was it the LT1763 that caused trouble on Urukul? Should we consider replacing it something more reliable like an LT3042? This would have the advantage of removing a BOM line.

Other than that, review complete. All looks good to me!

hartytp, 2019-10-26:
That comment about the LDO applies to +12V and +5V0A rails, but note the errata from Fastino.

Before changing, can you confirm that the LT1763 was indeed the part that caused issues on Urukul please?

Note that the review was only a look over the change log and not a full design review.

hartytp, 2019-10-26:
- [ ] Didn't you plan to replace SS16 with something rated to a higher voltage? Also, are you happy with the breakdown voltage of the FET. Since we don't have a TVS here and these components are before the EMI filter, we need to make sure that they can handle the maximum possible voltage spike (or just add a TVS to be safe!)

hartytp, 2019-10-26:
- [ ] Is the annotation about the OpAmps having JFET inputs true with the new opamps?

hartytp, 2019-10-26:
- [ ] are the dac output simulations updated for the new OpAmps?

hartytp, 2019-10-26:
- [ ] NB LT3042 needs 4.7uF min output capacitance. FWIW if we don't make this change we should, at the very least, increase IC16's output capacitance from 100nF to help ensure stability

hartytp, 2019-10-26:
Ok, now I'm done. :)

gkasprow, 2019-10-26:
The PoE rectifier uses 60V diodes. I can switch to 80V ones, but they have a [150mV higher](https://www.soemtron.org/downloads/disposals/ss14.pdf) voltage drop. However, PoE [spec](http://www.ieee802.org/3/poep_study/public/mar05/diamond_1_0305.pdf) says that max voltage is 57V, most switches use 48V though.

hartytp, 2019-10-26:
Ol if you think it’s fine then leave it as is

gkasprow, 2019-10-27:
>  Why is Ic1B1C DNP?

To lower the latency.

> 
> why is R124 undefined, not DNP?

I made it DNP. It's a placeholder.

> Is the annotation about the OpAmps having JFET inputs true with the new opamps?

They have 5pA of input bias, but they are not JFET. I changed the note.

I added 60V TVS as well.

gkasprow, 2019-10-27:
I also replaced the LT1763 LDOs with LT3042

closed 2019-10-27 by gkasprow

gkasprow, 2019-10-27:
new release
https://github.com/sinara-hw/Stabilizer/releases/tag/v1.1rc3

hartytp, 2019-10-27:
I can't see any issues with this design. Looks really lovely @gkasprow thanks again.