sinara-hw/Stabilizer #65: power supply rise time

gkasprow, 2019-11-27:
Once we switched to LT3042, the capacitors connected to the voltage setting pin cause the output voltage rise time of roughly 3s. This is not an issue that affects the functionality but looks weird for someone who powers the board for the first time. LEt's decrease the caps in next revision.
Another issue is related to PoE. If you unplug the RJ45 and plug it within a second the board won't work. It is an issue only with NETGEAR switch. With other switches, it works fine. It is caused by the fact that the FET that shorts the current limiting resistor is still on for some time after power disappears.

hartytp, 2019-11-27:
> If you unplug the RJ45 and plug it within a second the board won't work. It is an issue only with NETGEAR switch. With other switches, it works fine. It is caused by the fact that the FET that shorts the current limiting resistor is still on for some time after power disappears.

Aah, yes, that makes sense. In principle this could be fixed by adding a power present detection to the input and ORing that to the FET control line. I'm not sure it's worth the effort however.

Is it worth adding an annotation about this to the schematic? I'd argue that probably not because there are so many places in Sinara where power cycling within a second can cause issues. I think we probably need a high-level user guide/FAQ which points this out to users.

hartytp, 2019-11-27:
> Once we switched to LT3042, the capacitors connected to the voltage setting pin cause the output voltage rise time of roughly 3s. This is not an issue that affects the functionality but looks weird for someone who powers the board for the first time. LEt's decrease the caps in next revision.

:+1: This isn't an ultra-low design adn the cap doesn't need to be that large anyway.

marmeladapk, 2019-12-02:
@gkasprow You can also use PGFB pins to make rise time shorter.

gkasprow, 2019-12-02:
That's true. It is dual-use pin.

gkasprow, 2019-12-19:
The removal of the C136 22uFcapacitor solves the issue. I can cycle power as fast as I can and it works every time with the infamous Netgear switch.

hartytp, 2019-12-19:
Isn't it better to just use the normal fast start/PGFB?

Will this be an issue for other designs as well?

gkasprow, 2019-12-19:
I'm talking about PoE issue.
PGFB is right solution.

gkasprow, 2019-12-19:
This affects Thermostat as well.

hartytp, 2019-12-20:
Broke this out to two separate issues to avoid confusion

closed 2019-12-20 by hartytp