sinara-hw/Stabilizer #72: PoE vs. EEM power conflict

sbourdeauducq, 2020-03-19:
Currently the behavior of Stabilizer when installed in a Kasli crate is rather nasty:
* If you don't connect the EEM cable, the user needs to power it via PoE, which is less common than regular Ethernet. The power from Kasli cannot be used in a simple way.
* If you do connect the EEM cable, and the user later connect a PoE source, then Stabilizer powers Kasli and other EEMs, and may even also apply power in reverse to any plugged-in power supply adapter. In general, there is, in several places, insufficient current handling capability for that.

I propose:
* Add hardware so that the EEM port by default can only act as a current sink.
* Under MCU command, the EEM port may act as a current source.
* Add a current limit on the EEM port so that if power is inadvertently applied (e.g. due to a firmware problem) to a Kasli system through the Stabilizer EEM cable, the issue will be detected and no excessive current will flow anywhere.
* The current limit, when it triggers, should disconnect EEM power permanently and notify the MCU. EEM power can be re-established with a pulse (or I2C command or similar if we use a fancy IC and not discrete logic) coming from the MCU.

gkasprow, 2020-03-20:
This can be done easily with ideal diode (2 transistors + 2 resistors) opamp and a few lines of code.

gkasprow, 2020-03-20:
@sbourdeauducq the Stabilizer can also be powered by a barrel connector or Molex connector.

sbourdeauducq, 2020-03-20:
Okay, we probably want a diode there so it only acts as a current sink, e.g. when PoE is connected at the same time.

gkasprow, 2020-03-20:
OK

gkasprow, 2020-03-20:
I prefer ideal diodes over standard diodes :)

sbourdeauducq, 2020-03-20:
Sure :)

gkasprow, 2020-10-10:
I added an ideal diode circuit. It also enables powering the EEM from PoE on request.
[image]
I'm not sure if latched overcurrent protection in the case of firmware malfunction is really needed. It would complicate the design and cause additional confusion to the user.
The PoE will switch off after exceeding the allowable power something should break here.
I can also add a simple current monitoring circuit that will warn the CPU that we are exceeding the safe current operation - let's say 90% of the PoE module max current. The Stabilizer consumes 160mA of idle current. With Pounder it consumes 552mA. Let's warn the CPU when the EEM current exceeds 2A. 
One solution is to use the current sense amplifier like in the Booster. It adds 3$ to the BOM.
[image]

Another solution is to use 4BJTs and MOSFET RDSon - it's very low cost but not precise. It may also cause issues bypassing disabled FET.
[image]

gkasprow, 2020-10-13:
@sbourdeauducq are you happy with such solution?

sbourdeauducq, 2020-10-14:
> I'm not sure if latched overcurrent protection in the case of firmware malfunction is really needed. It would complicate the design and cause additional confusion to the user.
> The PoE will switch off after exceeding the allowable power, nothing should break here.

Okay, fair enough. Let's just go with the ideal diode + Force_EEM_Source circuit then, without additional current limit or monitoring.

gkasprow, 2020-10-18:
OK, done

closed 2020-10-18 by gkasprow