sinara-hw/Stabilizer #76: [Stabilizer v1.1] Ethernet auto-negotiation fails with PoE switch

ryan-summers, 2020-06-09:
When connecting the Stabilizer v1.1 ethernet port to a Netgera GS305P PoE ethernet switch, the auto-negotiation phase of the ethernet connection never completes. This phase is entirely managed by the LAN8942A PHY, so it is assumed this failure is due to signaling issues on the ethernet pair. Indication of the failure to complete is denoted by the PHY basic status register never indicating a complete auto-negotiation as well as the LED on the RJ45 housing.

When the ethernet cable is then disconnected from GS503P and reconnected to a D-Link DGS-1005D ethernet switch (not PoE), the green green and orange link LEDs on the RJ45 housing, which indicates a 100Mbps connection has been established.

jordens, 2020-07-15:
I suspect there is a broader issue with the Ethernet PHY. I don't know whether it's hardware or software. A few observations:

* On the Stabilizer v1.0 that I have here, originally the Ethernet PHY worked just fine and I could connect to my switch. Now the link initialization fails (with and without autonegotiation, with and without PoE). The CPU talks fine with the PHY but the registers indicate link link down most of the time and then intermittent flashes on the LEDs and very short intervals of a other values in the registers. Now that Stabilizer doesn't even want to accept PoE from the switch (a Cisco Catalyst). The switch doesn't see anything worth noting.
* On a Stabilizer v1.1 from us that's currently with Technosystem for repair, the failure involving the hot CPU started suddenly. TS suspects a short between NRST and RMII TXD.
* On the other Stabilizer v1.1 PoE works but the link with the Cisco switch doesn't come up either.

I tried poking around a bit on the v1.0 board and probing a few things but didn't see anything that caught my eye.

@gkasprow

gkasprow, 2020-10-10:
I don't remember exactly, But once I replaced broken PHY on either Stabilizer or Thermostat. What's the status of this issue?

jordens, 2020-10-11:
No news. Let's try gathering some data on this: @airwoodix @hartytp @nkuh have you seen anything like this?

hartytp, 2020-10-11:
@jordens this is the only somewhat relevant issue I'm aware of us having seen https://github.com/sinara-hw/Stabilizer/issues/47

gkasprow, 2020-10-11:
Thanks for reminding me that I replaced it on Thermostat.
It's the same PHY and CPU, but they are supplied from different SMPS.

nkuh, 2020-10-12:
> 
> 
> No news. Let's try gathering some data on this: @airwoodix @hartytp @nkuh have you seen anything like this?

We don't use PoE.

gkasprow, 2020-10-18:
the same PoE circuit is used in Thermostat design. I tested it with Netgear GS305P and it works fine.
The Stabilizer I have has broken PoE circuit - there is an internal short circuit that is hard to locate. The other v1.1 boards may have a similar issue with shorted traces. I think we cannot do much here.

closed 2020-10-18 by gkasprow