sinara-hw/Stabilizer #83: DAC updates may cause transient spikes at specific code boundaries

ryan-summers, 2020-10-19:
When the output DAC on stabilizer crosses the 10-bit code boundary, there is a ~20mV spike on the output waveform. For example, the following output was captured when the DAC code was repeatedly transitioned from 0x83FF to 0x8400 at a rate of 50KHz. The directionality of the spike appears to be a function of the transition direction (e.g. transitioning to a lower code will cause a positive spike):
[image]

ryan-summers, 2020-10-19:
Reference https://github.com/quartiq/stabilizer/issues/98 for further investigation and background information

ryan-summers, 2020-10-19:
Note that this appears to be in-spec for the AD5542A and is specified in Figure 20: [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/AD5512A_5542A.pdf)
[image]

jordens, 2020-10-19:
Let's

* Check whether fastino or stabilizer v1.0 are affected
* Copy/harmonize the afe with fastino
* Determine if this is dac batch dependent (w.r.t fastino and v1.0)

jordens, 2020-10-19:
This is assuming the datasheet axes are µs and not ns which is likely. The glitch is 1.1 nVs typ = 11 mV*0.1 µs consistent with that figure.

jordens, 2021-10-09:
There havent been any suggestions to resolve it or complaints about it. Seems inherent to the DAC architecture.

closed 2021-10-09 by jordens