sinara-hw/Stabilizer #86: digital outputs

jbqubit, 2020-12-08:
In v1.1  MCX DI0 and DI1 are input only. Consider making them software reconfigurable input or output like on EEM DIO. https://github.com/quartiq/stabilizer/issues/86#issuecomment-735692391

jordens, 2020-12-08:
Pleas don't do that as long as it introduces incompatibilities and as long as there are unaddressed impacts on the firmware.

dtcallcock, 2020-12-11:
For background, not being able to trigger a scope when scanning the output is the first use case I've thought of.  If Stabilizer is to become [some kind of lab swiss army knife](https://github.com/quartiq/stabilizer/issues/147) as envisaged, then not having a simple way of getting a digital output seems short-sighted, and I think should at least be considered at the v1.2 stage.

I see a few options:

- Add biderectionality to MCX DI0 and DI1
  - Make them software reconfigurable as @jbqubit suggests above.  Introduces incompatibilities - presumably tractable ones but at a cost.
  - Make them DIP switch configurable.  Perhaps easier to support backward compatibility in firmware as a v1.1 board just looks like a v1.2 board with the DIP switches set to input (conversely v1.2 boards would not require this feature to be supported in firmware right away).
- Add a header or a couple of vertical MCX to the board and use pigtails to break out (assuming there is space and the free uC pins).
- Use the EEM connector.  This is not as straightforward as one would like but there are a few ways you could imagine doing it:
  - Use stock Stabilizer + DIO board and a custom ribbon cable.  A custom ribbon cable which switches pin locations is required as the input/output direction config on Stabilizer doesn't match the banks of the DIO boards.
  - Use a Stabilizer population variant + stock DIO board and standard ribbon cable.  With a population variant, the EEM directions could be made to match DIO banks (ie. 4x input and 4x output).
  - Make a new DIO_EEM with per channel direction selectivity (non-isolated).  I think such a simple board has been discussed in the past, but I'm guessing the upcoming shift to CPCIS makes this less appealing.
- We decide I'm overstating the need so do nothing.  Note that the EEM route above doesn't require any changes to v1.2 so is still open down the line.

gkasprow, 2020-12-29:
let's leave it for next release.

gkasprow, 2021-10-09:
The Stabilizer was designed in a way that it can act as either an EEM upstream or downstream device. So one can connect DIO module.
However, in many cases just one output may be sufficient, so we can replace the input buffers with bi-dir ones.
Let's add the pulldown resistor so the default configuration will be compatible with 1.2 firmware.
I wouldn't make it configurable by a dip switch, because we need to modify the MCU direction anyway. Let's define it by software.

closed 2021-10-09 by gkasprow