sinara-hw/Stabilizer #87: P12VA LED occasionally does not illuminate

ryan-summers, 2021-01-06:
On my V1.1 stabilizer board, I have observed cases when the P12VA LED does not illuminate even after the board remains powered for a number of minutes. Probing the pin with a DMM indicates a voltage of ~0.6V.

Removing and reapplying power is sufficient to get P12VA's LED to illuminate again and probing indicates correct P12VA voltage.

Is this a known issue with v1.1 variants?

dnadlinger, 2021-01-06:
No idea about the issue, and haven't checked the schematics, but this sounds exactly like the classic regulator reverse biasing issue.

gkasprow, 2021-01-07:
the LDOs have protection diodes that should prevent such issues.
Do you have any mezzanine plugged in?
Can you check if  IC13 gets hot?

ryan-summers, 2021-01-08:
No mezzanine, IC13 remains cool the whole time. Here's a video of an occurrence (it can be repeated somewhat reliably, such as 1 boot out of 5-10?)

https://user-images.githubusercontent.com/8771450/104006880-66bd4380-51a7-11eb-80e6-5faf0f61f964.mp4

ryan-summers, 2021-01-08:
Probing Pins 1/2 (VIN) of IC13 indicates 12.96V and probing pin 3 (EN) indicates 5.737V

ryan-summers, 2021-01-08:
The LT3042 data sheet mentions:
```
A parasitic substrate diode exists between OUT and GND pins of the LT3042; do not drive OUT more than 0.3V below GND during normal operation or during a fault condition.
```

However, measuring OUT indicates -0.62 V

hartytp, 2021-01-08:
as @dnadlinger says, it sounds like the LDO needs a reverse protection diode (or if there is one then it's not doing its job)

dnadlinger, 2021-01-08:
Looks like v1.1 had a SMBJ12A on the 12V0A rail rather than a low-forward-voltage Schottky diode – try soldering in an SS16, DB2S20500L, or something like that, which should fix the issue. v1.2 has an SS16 instead.

dnadlinger, 2021-01-08:
(Also, in retrospect, I think I have seen this issue once or twice, though I blamed it on a questionable power supply at the time.)

gkasprow, 2021-04-15:
it was fixed in rev1.2 already

closed 2021-04-15 by gkasprow