sinara-hw/Stabilizer #89: wronf REV ID

gkasprow, 2021-01-23:
Stabilizer v1.2 still has REV ID =1

sbourdeauducq, 2021-01-25:
Did you send any v1.2 boards to production with this problem, or will it get fixed before?

sbourdeauducq, 2021-01-25:
Well I guess that it'll get fixed since this is simply a resistor setting.
What's the correct value?

gkasprow, 2021-01-25:
it should be "2"
I don't know if that was already produced.
@kaolpr ?

gkasprow, 2021-04-15:
CTI produced a batch of it.
@akaminska small fix needs to be done after the production.

gkasprow, 2021-04-15:
the resistors can be removed and two shorts need to be installed
[image]

gkasprow, 2021-10-09:
changed to 3 since we are adding one CPU pin function in rev1.3

closed 2021-10-09 by gkasprow