sinara-hw/Stabilizer #91: DI0/1 jitter

jordens, 2021-02-07:
I would like to use DI0/1 for accurate and low-jitter timestamping and limiting. Currently it's a plain digital buffer driven from 3V3 digital and limited by the stability of that power supply, without hysteresis, requiring 1V/10ns slew rate.

Maybe we can give it a better supply/threshold and consider a 1V-threshold Schmitt trigger.
And we should add a pull-down.

I don't think this should become a zero-crossing detector. That wold make it useless for digital inputs.

gkasprow, 2021-10-09:
@jordens what about such a circuit?
I added TVS protection, output buffer that can be disabled, Schmitt input.
[image]

gkasprow, 2021-10-09:
TVS absorbs the energy. It gives a chance to the next chip to survive. 51R is to reduce the current in case ESD was really high.