sinara-hw/Stabilizer #94: SMA/MCX connector pins are too long

SingularitySurfer, 2021-03-23:
[image]
When inserting Stabilizer 1.2 next to a panel with RF gasket, the pins damage the gasket on insertion and continue to be shorted when Stabilizer is inserted all the way. Kapton tape over the pins gets punctured. Clipping off the pins and then covering them with kapton fixes the issue but the sharp edges from cutting might eventually still cut through.

gkasprow, 2021-03-23:
@akaminska @kaolpr you should trim the pins before assembling the panels

jordens, 2021-03-23:
The SMAs should be migrated to the longer ones you found.
Maybe also longer ones are available for MCX.

jordens, 2021-10-08:
AFAICT all DIO-SMA, IDC-SMA, Phaser, Urukul, Mirny, Kasli, Stabilizer, Pounder, Clocker, are affected. Phaser is supposed to use the longer expensive SMA connectors currently. Low-cost alternatives for that and application to all SMA boards to move the pins away from the RF shield lip would be great.
On Stabilizer and other MCX boards I think this also affects the MCX connectors. Likely there is no long MCX connector.

gkasprow, 2021-10-09:
By ring I mean the chamfer
[image]