sinara-hw/Stabilizer #96: Frontpanel cut-outs/silkscreen for standalone enclosure

SingularitySurfer, 2021-07-12:
It would be great to have the option for a standalone enclosure for Stabilizer.

Stabilizer fits nicely into [these](https://www.digikey.de/product-detail/en/hammond-manufacturing/1455L1602/HM897-ND/1090709), but the plastic end-caps need cut-outs for connectors and silkscreen labels would be great too.

Can we adapt the usual front-panel drawings for these end-caps?

gkasprow, 2021-07-12:
You can ask Creotech, they can do that.

jordens, 2021-07-12:
It think it doesn't completely fit because the board is a bit longer than 160 mm due to the SMA PCB protrusion towards the panel. 

IMO for all Sinara boards we should aim to just reduce the board length at the back by a couple mm to remain within the standard 160 mm. Since we don't use the orthogonal backplane anywhere we don't need to meet that absolute board end location.

SingularitySurfer, 2021-07-12:
I think the extra 3 mm should not be a big issue for this case since the end-caps usually reach about 6-7 mm around the outside. It would just lead to some pressure on the board.

jordens, 2021-10-08:
[These](https://www.digikey.de/product-detail/de/hammond-manufacturing/1455L1602/HM897-ND/1090709) don't. You have 160 mm between the inner cap faces.

closed 2023-06-22 by maciejprzybysz