sinara-hw/Stabilizer #98: pull downs on DAC LDAC and CLR

jordens, 2021-10-09:
https://github.com/quartiq/stabilizer/issues/447

@gkasprow can we add this for v1.3 still?

gkasprow, 2021-10-09:
added pullup
[image]

gkasprow, 2021-10-09:
edit, wrong image

gkasprow, 2022-01-12:
@jordens we can solve the issue completely by allowing the CPU to turn on the output amplifier power after the DAC was properly reset/initialized. We can also add a delay to the +/-12V rail LDOs. 1s would be fine.
Moreover, the AD5542ABCPZ is unavailable. We can install AD5542ARZ which does not have CLR input. So the CPU would have to send SPI command first, then turn on the AFE supply.

jordens, 2022-01-12:
Ack. Ensuring the CPU has to explicitly enable AFE power is fine. Then we also don't need the CLR input.

gkasprow, 2022-01-12:
Even with CLR input, we must wait for the CPU to boot, then toggle the CLR H->L which takes time during which the output of the DAC is negative.

jordens, 2022-01-12:
A pull up on CLR that would passively clear the DAC to mid-scale (0 V).

gkasprow, 2022-01-12:
The pullup is already there for some time. It is falling edge so I'm not that sure if the issue is completely solved.

jordens, 2022-01-12:
There is a power-on reset ensuring valid initial DAC register. The pull-up on CLR then prevents any LDAC/SPI activity from loading invalid shift register data into it. In that aspect CLR is level sensitive.

jordens, 2022-01-12:
And AFAICT there has not been a production run with that pull up yet.

jordens, 2022-01-12:
Ah. On CLR it should be a **pull-down**. Right @svrotter ?

gkasprow, 2022-01-12:
The v1.3 was produced already. Anyway, these DACs have a one-year lead time so we have to use alternatives without CLR.

gkasprow, 2022-01-12:
So, we must keep LDAC high to make sure the random register value is not loaded to DAC and pull CLR to GND to generate H to L toggle and clear the DAC. I will also add a signal that turns the +/-12V supplies on.

jordens, 2022-01-12:
Last I heard production was still being held up by various chips missing. But you know more.

Agreed. I don't think delaying the CLR assertion (by some late pulldown after an early pullup) and LDAC deassertion is necessary or useful. I'd just put the right pulldown/pullup and trust that the DAC resets itself and then ignores whatever is going on on LDAC/SPI. If in addition we have that AFE power enable (with pull-disable) then we are certainly safe.

svrotter, 2022-01-13:
For me, the most important thing was to patch in a pull up on !LDAC because it seemed that the DAC outputs garbage at power-reset for some time with Stabilizer v1.2 (see https://github.com/quartiq/stabilizer/issues/447). 
The DAC itself powers up at midscale without even reading any registers when !LDAC is high and for every state of !CLR.
This is also described in the POWER-ON RESET section of the data sheet (except the !CLR statement, but I'll come to that later).

@jordens said:
>There is a power-on reset ensuring valid initial DAC register. The pull-up on CLR then prevents any LDAC/SPI activity from loading invalid shift register data into it. In that aspect CLR is level sensitive.

1. you mean pull down on !CLR
1. in one important detail this is not true. A pull down on !CLR results in ignoring !LDAC **pulses**. When !LDAC is constantly low, it just keeps the output latching and pushing out garbage on power up

@jordens said:
>Ah. On CLR it should be a pull-down. Right @svrotter ?

The !CLR is edge sensitive, so pull resistors at this line in any direction won't really solve the problem. Clearing should be issued after the DAC is fully powered, which is when the pulses we were talking about already occurred.

@gkasprow 
>So, we must keep LDAC high to make sure the random register value is not loaded to DAC and pull CLR to GND to generate H to L toggle and clear the DAC. I will also add a signal that turns the +/-12V supplies on.

That should be correct. I'd do the following start up sequence:
1. keep !LDAC high with an external pullup
1. generate a falling edge on an externally pulled up !CLR
1. write valid data to the serial register (microcontroller sending midscale value)
1. set an externally pulled low AFE supply enable line high
1. release !CLR
1. set !LDAC low

BTW: having the microcontroller to turn on the AFE explicitly is very useful in any case in my opinion.

jordens, 2022-01-13:
Ok. But then 2., 5., 6. can be skipped (assuming there is a pullup on CLR). What's the point?

svrotter, 2022-01-13:
(2) and (5) could be skipped if there is a pull up on !CLR **and** if (3) is executed correctly. You could say that it's another level of secure start up for me.
(6) has to be executed in order to make the DAC work like it is intended right now (apply every new incoming data as soon as possible)

gkasprow, 2022-01-13:
The new revision has CLRn pulled low
[image]
LDAC pulled high
[image]
Assembly option with AFE power turned on by CPU by default. There is an option to remove 2 resistors and install another one and have AFE turned on automatically at the end of power-up sequence.

svrotter, 2022-01-13:
>The new revision has CLRn pulled low

As long as !LDAC is pulled high, this works. It's basically the configuration on my patched v1.2. 
If you want to clear the DAC register you just have to set it high first and then generate the falling edge with the uC.
I just experienced that you get the least problems with weird start up behaviour when these kind of lines are pulled to their inactive state (eg. active high signals have a pull down and active low signals have a pull up resistor)

> Assembly option with AFE power turned on by CPU by default. There is an option to remove 2 resistors and install another one and have AFE turned on automatically at the end of power-up sequence.

Is there a missing image? If not I'm not sure whether I understand what you're referring to

gkasprow, 2022-01-13:
[image]

svrotter, 2022-01-13:
Now I understand what you mean. Looks good to me

closed 2023-06-22 by maciejprzybysz