sinara-hw/Stabilizer #113: v1.3.4 changelog

maciejprzybysz, 2023-06-22:
- [x] #112 
- [x] SMA changed to DELOCK 65848
- [x] SMA/MCX to panel distance changed for uniformity with other boards

jordens, 2023-06-22:
We've had a report where the threaded part of a few DELOCK 65848 SMA connectors loosened itself from the square body when tightened with a torque wrench (to presumably 1 Nm). It would be very annoying if the connector doesn't survive even a few mating cycles with a torque wrend. I know proper tightening can be tricky, especially given the connectors density on our front panels. Could you double check that the connector survives 2 Nm torque? I don't have these laying around.

jordens, 2023-06-26:
@maciejprzybysz could you please check the above?

maciejprzybysz, 2023-06-26:
@jordens We are going to check this, but we need few more days to close ongoing tasks.

pkozakiewicz, 2023-06-29:
@jordens We checked durability of both Delock and Amphenol SMA connectors used in the past. None of them survived more than 1,2Nm. Are you able to identify the ID of the affected device?

jordens, 2023-06-29:
1 or 2 Nm? Failures with 1 Nm would be disappointing as that's "the" recommended torque and what a typical torque wrenche will do in practice.

@pkozakiewicz ID as in serial number?

jordens, 2023-06-29:
The ones where we noticed it was Creotech, recent Urukuls.
But you are right. The [recommended torque (Amphenol)](https://web.archive.org/web/20100331083020/http://www.amphenolrf.com/products/CatalogPages/SMA.pdf) for brass is 0.45 Nm, max 0.6 Nm. So they have an empirical safety factor of 2.