sinara-hw/Stabilizer #119: v1.3.2 DAC8831IBD noise

MorganTL, 2024-07-18:
The DAC output is very noisy on both channel when using DAC8831IBD. The issue is fixed after replacing the DAC8831IBD with AD5542ABCPZ-REEL7 

## Test setup
- Connect 1Vpp sine wave to ADC and oscilloscope blue channel using T splitter
- Connect DAC to oscilloscope yellow channel
- With the [ff2c2c1](https://github.com/quartiq/stabilizer/tree/ff2c2c19dce6489de60693c55d5d919c17dd17a0) firmware from [quartiq/stabilizer](https://github.com/quartiq/stabilizer)

### Result With DAC8831IBD 
[image]

### Result after replacing DAC8831IBD with AD5542ABCPZ-REEL7   
[image]

### Related issue
#107

gkasprow, 2024-07-18:
DAC8831IBD was used due to component shortages

gkasprow, 2024-07-18:
Noise may be caused also by SPI timing issues

gkasprow, 2024-07-31:
it passes our automated test. But the test is not very detailed and checks mainly assembly

dnadlinger, 2026-11-11:
I am also seeing glitches on all the ones with the TI DAC I tested (and on none of the ADA5542A ones), and they look exactly like what I would expect from DAC glitches (bit pattern dependence, etc.).

The DAC8831's datasheet glitch spec is considerably worse than the ADA5542A's (35 nV·s vs 1.1 nV·s), and this could already be enough to be problematic (though, without doing a precise measurement, I'm measuring about 2–3x that).

@gkasprow: Based on this, I don't think DAC8831 should be considered a viable replacement for the ADA5542A.