sinara-hw/Stabilizer #121: Trouble Building Firmware and Extracting Application

kvziegler, 2024-08-02:
After cloning the stabilizer repo per the instructions, https://quartiq.de/stabilizer/setup.html, the cargo build --release fails. The error provided seems to point to the zlib1g dependency.

Error:
linking with `rust-lld` failed: exit status: 127
note: rust-lld: error while loading shared libraries: libz.so.1: cannot open shared object file: No such file or directory

error: could not compile `stabilizer` (bin "lockin") due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: linking with `rust-lld` failed: exit status: 127

The versions of zlib1g and dfu-utils are as follows:
zlib1g: 1:1.2.11.dfsg-2ubuntu9.2 amd64
dfu-util: 0.9-1

jordens, 2024-08-03:
That sounds like a broken rust compiler installation or your OS is missing that library, nothing to do with Stabilizer or this firmware.

Maybe you mixed up x86_64 and 32 bit architectures. Or the library is just missing.

jordens, 2024-08-03:
And this is most definitely not a hardware issue suitable for this repo.

jordens, 2024-08-03:
Is nix involved? Which OS is this?
How did you install rust?

kvziegler, 2024-08-05:
My apologies, can you point me towards the correct repo suitable for these kinds of software issues. I will continue this there.

Otherwise, Nix is involved. I created a flake.nix exactly as described by the official M-Labs manual. I am using the Nix package manager (version 2.22.0) and my system runs on Ubuntu (22.04).

I installed rust via https://rustup.rs/

gkasprow, 2024-08-05:
This is HW repo. Use SW repo https://github.com/quartiq/stabilizer

jordens, 2024-08-05:
What do you mean with "official m-labs manual"? The manual you cited above does not involve any nix stuff. It's not needed or helpful here. You're issue looks like one of the nix issues you'll find online.

closed 2024-08-05 by jordens